ros-controls/ros2_controllers #1167: Ackermann preceding reference command interfaces are not correct

wittenator, 2024-06-09:
**Describe the bug**
I am building a ackermann-controlled vehicle and would like to control the mobile base via steering angle and linear velocity instead of a twist since I have a preceding pure pursuit controller. In the documentation ([AckermannDrive](https://control.ros.org/iron/doc/ros2_controllers/steering_controllers_library/doc/userdoc.html#references-from-a-preceding-controller) it says that the references are linear velocity and angular position of the steering. Setting these references via ForwardController reveals that the angular position command interface is in fact a angular velocity command interface. Since the steering library is worked on rn anyway: Would it be an idea to let people switch between controlling the vehicle via Twist or AckermannDrive messages? This would remove the slightly awkward setup of having to separate ForwardCommand controllers or a completely self-written controller just to set steering angle and velocity instead of a Twist.

**To Reproduce**
Steps to reproduce the behavior:
1. Start an Ackermann vehicle
2. Attach two forward command controllers to the Ackermann controller e.g.
 ```forward_steering_controller:
  ros__parameters:
    joints:
      - ackermann_steering_controller/angular

forward_velocity_controller:
  ros__parameters:
    joints:
      - ackermann_steering_controller/linear 
```
4. Set a steering command and a linear velocity command

**Expected behavior**
Setting a steering angle should be passed on to the angle commands and not be correlated with the velocity.

**Screenshots**
[image]
The blue line is the commanded steering angle and the orange one is the true steering angle. Changing the linear velocity also influences the steering angle.

**Environment (please complete the following information):**
 - OS: Ubuntu 22
 - Version Iron

**Additional context**
Add any other context about the problem here, especially include any modifications to any ros2_controllers that relate to this issue.

christophfroehlich, 2024-06-09:
You are right, the reference interfaces have the same meaning than body twist -> #1168 

If you have your own controller, why do you need the ackermann steering ctrl at all? Just for odometry calculations?

The author of the lib mentioned [support ackermann_msg reference messages](https://github.com/ros-controls/ros2_controllers/pull/484) as missing feature. You want to implement this and submit a PR? 

There are other issues with this controller, see #1150. As you are working with it currently, maybe you can have a look and review it?

wittenator, 2024-06-09:
We have two controllers mostly for separation of concerns reasons and for the odometry that comes for free (also I want to push the usage of ros2_control and ros2_controllers internally and maybe push people to migrate their controllers like pure pursuit and MPCs to ros2_control :) ). I already started work implementing a switch parameter in the steering library that allows switching between Twist and Ackermann drive. I forked the repo and branched off #1150 for this, so I think that I'll get some experiences with the new changes. I'll report once I have something to useful to say (I just have to migrate our project to Jazzy first)^^ But thanks for the clarification already!

wittenator, 2024-06-17:
For reference: I added a WIP PR here in #1171 . I still need to add tests for the new code though, but our simulated car drives very well with the changes. I will add a video to the PR as well.

PR #1168 (2024-06-18): [Steering controllers library] Reference interfaces are body twist
Fixes #1167 

I'm not sure if the intention of the author was to have position input, but as it is implemented we have a body twist input on reference interfaces (velocity, not position). This PR fixes this, without any attempt to deprecate the old behavior because it was just wrong.

Tests succeed on iron distro
```
$ colcon test --packages-select steering_controllers_library ackermann_steering_controller bicycle_steering_controller tricycl
e_steering_controller
Starting >>> steering_controllers_library
Finished <<< steering_c [... cut]
PR #1173 (2024-06-19): [Steering controllers library] Reference interfaces are body twist (backport #1168)
Fixes #1167 

I'm not sure if the intention of the author was to have position input, but as it is implemented we have a body twist input on reference interfaces (velocity, not position). This PR fixes this, without any attempt to deprecate the old behavior because it was just wrong.

Tests succeed on iron distro
```
$ colcon test --packages-select steering_controllers_library ackermann_steering_controller bicycle_steering_controller tricycl
e_steering_controller
Starting >>> steering_controllers_library
Finish [... cut]
PR #1174 (2024-06-19): [Steering controllers library] Reference interfaces are body twist (backport #1168)
Fixes #1167 

I'm not sure if the intention of the author was to have position input, but as it is implemented we have a body twist input on reference interfaces (velocity, not position). This PR fixes this, without any attempt to deprecate the old behavior because it was just wrong.

Tests succeed on iron distro
```
$ colcon test --packages-select steering_controllers_library ackermann_steering_controller bicycle_steering_controller tricycl
e_steering_controller
Starting >>> steering_controllers_library
Finish [... cut]
closed 2024-06-18 by bmagyar