ros-controls/ros2_controllers #1175: Admittance controller - Endeffector frames that are not part of the urdf chain are not accepted

firesurfer, 2024-06-19:
**Describe the bug**

On a typical robot arm we have a kinematic chain going from the base up to tool frame. 
The endeffector(s) and their frames are then connected to that tool frame. 

When we try to select an endeffector frame that is not part of the kinematic chain but connected to it we get such an error:

```
[ros2_control_node-14] [ERROR] [1718797580.334696574] [kinematics_interface_kdl]: The link ur_top/my_endeffector/attach was not found in the robot chain. Available links are: 
[ros2_control_node-14] base_link
[ros2_control_node-14] ur_top/base_link
[ros2_control_node-14] ur_top/base_link_inertia
[ros2_control_node-14] ur_top/shoulder_link
[ros2_control_node-14] ur_top/upper_arm_link
[ros2_control_node-14] ur_top/forearm_link
[ros2_control_node-14] ur_top/wrist_1_link
[ros2_control_node-14] ur_top/wrist_2_link
[ros2_control_node-14] ur_top/wrist_3_link
[ros2_control_node-14] ur_top/flange
[ros2_control_node-14] ur_top/tool0

```

**To Reproduce**

Select a frame behind the kinematic chain as `control.frame.id`

**Expected behavior**

Endeffector frames that are connected to the kinematic chain should be selectable as `control.frame.id`

**Environment (please complete the following information):**

 - Version: Iron

**Additional context**

When we take a look at the example from the documentation we see that this case has been initially thought of by the `external` parameter. This parameter is not used anymore:

```
    control:
      frame:
        id: tool0 # Admittance calcs (displacement etc) are done in this frame. Usually the tool or end-effector
        external: false # control frame exists within URDF kinematic chain
``

saikishor, 2024-06-19:
Hello @firesurfer!

Ofcourse, this is pretty much how it works. Your and effector and its frames information is expected to be in the robot description so that the controller can get it from there. Any external frames are expected to fail as you have mentioned.

Thank you!

firesurfer, 2024-06-19:
Well they are in the urdf description but not in the chain from: 

`kinematics.base` to `kinematics.tip` 
The reason I have it this way is that I want to change the "admittant" frame at runtime.

firesurfer, 2024-06-19:
But nevermind. I managed to work around the issue by:

1. Deactivating the controller
2. Set `kinematics.tip` and `control.frame.id` 
3. Configure the controller again (As explained in: https://github.com/ros-controls/ros2_control/issues/1033)

closed 2024-06-20 by saikishor