ros-controls/ros2_controllers #1183: How to send repeated trajectory points to the pid_position_controller to make the joint rotate continuously

HPCLOL, 2024-06-27:
I use ros2 Iron to simulate the robot in gazebo classic. 
﻿
Now, I am trying to make a joint rotate continuously. For some reasons, I have to control it with pid_position_controller in https://control.ros.org/rolling/doc/gazebo_ros2_control/doc/index.html#using-pid-control-joints. During the first rotation of the joint, I send it a series of trajectory points from -180 to 180 degrees. But because the pid_position_controller needs to receive increasing angle commands, I have to send trajectory points from 180 to 540 degrees to the controller so that it can complete the second rotation. If I still send trajectory points from -180 to 180, the joint will quickly rotate a circle in the opposite direction and then execute my trajectory, which sometimes causes the model to crash.
﻿
Is there any way I can send repeated trajectory point commands to achieve continuous rotation of the joint?
﻿
Thank you so much.

christophfroehlich, 2024-06-27:
Is there a chance to use a velocity command?

christophfroehlich, 2024-06-27:
I think that a continuous URDF joint is not the best choice for a multi-turn joint. Why can't you use just a revolute joint, if the position is important for more than one rotation?

HPCLOL, 2024-06-27:
>Is there a chance to use a velocity command?

Using velocity commands in my application scenario may be a very inconvenient thing.

> I think that a continuous URDF joint is not the best choice for a multi-turn joint. Why can't you use just a revolute joint, if the position is important for more than one rotation?

Actually the joint type I used is revolute, but I set the upper and lower limits so large that it can rotate continuously when I send it increasing position commands. I don't quite understand what you mean, even if I use a revolute joint it doesn't seem to solve the problem of crossing the boundary angles. When the angle of the joint trajectory point command exceeds the upper and lower limits, the joint cannot move normally.

christophfroehlich, 2024-06-27:
If it is a revolute joint, you have to send increasing values.

A continuous joint doesn't have position limits, and therefore we implemented this angle_wraparound in PID controller and JTC: It will always use the shortest distance on the unit cycle. This is somehow the same behavior as you mentioned.
 
The pid controlled joints in gazebo_ros2_control don't have this feature. But you can use the [PID controller](https://control.ros.org/rolling/doc/ros2_controllers/pid_controller/doc/userdoc.html#parameters) instead, and try the angle_wraparound.

HPCLOL, 2024-06-27:
Ok, I will try it out, thank you very much!

closed 2025-05-12 by github-actions[bot]