ros-controls/ros2_controllers #1197: Some problems with config files when I use ros2_canopen in combination with diff_drive_controller, the terminal output is "command interface already in available list. This can happen due to multiple calls to 'configure'".

15906185391, 2024-07-08:
**Describe the problems**
Hi,

I'm a student and I'm trying to use diff_drive_controller with ros2_canopen, I followed the tutorials about ros2_canopen and ros2_control, I referenced the config files the sample program from github.

Problem:
When I run the program, ros2 diff_drive_controller and ros2_canoepn work fine respectively. When I call 
```bash
ros2 topic pub --rate 30 /diffbot_base_controller/cmd_vel geometry_msgs/msg/TwistStamped "
twist:
  linear:
    x: 0.7
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 1.0"
```
the robot in rviz turned normally, but wheels that ros2_canopen drives are not turning. Some of the files are not configured correctly, but I don't know how to configure them correctly.

the terminal output is 
```bash
[ros2_control_node-3] [WARN] [1720421078.777601620] [resource_manager]: (hardware 'diff_robot'): 'left_wheel_joint/position' state interface already in available list. This can happen due to multiple calls to 'configure'
[ros2_control_node-3] [WARN] [1720421078.777606030] [resource_manager]: (hardware 'diff_robot'): 'left_wheel_joint/velocity' state interface already in available list. This can happen due to multiple calls to 'configure'
[ros2_control_node-3] [WARN] [1720421078.777608982] [resource_manager]: (hardware 'diff_robot'): 'right_wheel_joint/position' state interface already in available list. This can happen due to multiple calls to 'configure'
[ros2_control_node-3] [WARN] [1720421078.777611514] [resource_manager]: (hardware 'diff_robot'): 'right_wheel_joint/velocity' state interface already in available list. This can happen due to multiple calls to 'configure'
[ros2_control_node-3] [WARN] [1720421078.777615605] [resource_manager]: (hardware 'diff_robot'): 'left_wheel_joint/position' command interface already in available list. This can happen due to multiple calls to 'configure'
[ros2_control_node-3] [WARN] [1720421078.777618048] [resource_manager]: (hardware 'diff_robot'): 'left_wheel_joint/velocity' command interface already in available list. This can happen due to multiple calls to 'configure'
[ros2_control_node-3] [WARN] [1720421078.777620589] [resource_manager]: (hardware 'diff_robot'): 'right_wheel_joint/position' command interface already in available list. This can happen due to multiple calls to 'configure'
[ros2_control_node-3] [WARN] [1720421078.777622945] [resource_manager]: (hardware 'diff_robot'): 'right_wheel_joint/velocity' command interface already in available list. This can happen due to multiple calls to 'configure'
```

Here are my config files. 

diffbot.urdf.xacro
```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://wiki.ros.org/xacro" name="test_robot">
    <xacro:include filename="$(find canopen_tests)/urdf/robot_controller/diffbot_description.urdf.xacro"/>
    <xacro:include filename="$(find canopen_tests)/urdf/robot_controller/diffbot.ros2_control.xacro"/>
    <!-- Import Rviz colors -->
    <xacro:include filename="$(find ros2_control_demo_description)/diffbot/urdf/diffbot.materials.xacro" />
    
    <xacro:diffbot prefix="">
    </xacro:diffbot>

    <xacro:diffbot_ros2_control
        name="diff_robot"
        prefix=""
        bus_config="$(find canopen_tests)/config/robot_control/bus.yml"
        master_config="$(find canopen_tests)/config/robot_control/master.dcf"
        can_interface_name="vcan0"
        master_bin="" />
</robot>

```
diffbot.ros2_control.xacro

```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:macro name="diffbot_ros2_control" params="
      name
      prefix
      bus_config
      master_config
      can_interface_name
      master_bin
      ">
[... 607 lines cut ...]
/slave_node_1/get_transition_graph
/slave_node_1/get_type_description
/slave_node_1/list_parameters
/slave_node_1/set_parameters
/slave_node_1/set_parameters_atomically
/slave_node_2/change_state
/slave_node_2/describe_parameters
/slave_node_2/get_available_states
/slave_node_2/get_available_transitions
/slave_node_2/get_parameter_types
/slave_node_2/get_parameters
/slave_node_2/get_state
/slave_node_2/get_transition_graph
/slave_node_2/get_type_description
/slave_node_2/list_parameters
/slave_node_2/set_parameters
/slave_node_2/set_parameters_atomically
/transform_listener_impl_5e41e8bf69c0/get_type_description
```

**Environment :**
 - OS: Ubuntu 22.04
 - Version iron

I tried many times but I can not config it correctly. Could you please help me?

christophfroehlich, 2024-07-12:
I don't know ros2_canopen, but what I see now: 
* a remapping on the wrong place (remapping of cmd_vel for robot_state_publisher, but there is no such a topic there).
* you configure two controllers for the same interfaces? What is robot_controller for? maybe ros2_canopen does not work with diff_drive controller if it needs special interfaces.

15906185391, 2024-07-13:
> I don't know ros2_canopen, but what I see now:
> * a remapping on the wrong place (remapping of cmd_vel for robot_state_publisher, but there is no such a topic there).
> * you configure two controllers [...]

Thanks for the reply! 

I followed the tutorial: https://ros-industrial.github.io/ros2_canopen/manual/rolling/user-guide/how-to-create-a-robot-system.html

Maybe I configure two controllers for the same interfaces. I commented out robot_controller and still have the same problem. ros2_canopen is configured with state_interface and command_interface, and diff_controller should have been configured once as well, and I'm trying to suppress one of the controllers for state and command interface configurations.

christophfroehlich, 2024-07-17:
Sorry, I have no experience with ros2_canopen. Maybe you should ask there, and link this issue here for reference.

15906185391, 2024-07-18:
> Sorry, I have no experience with ros2_canopen. Maybe you should ask there, and link this issue here for reference.

Thanks for your reply! This is a ros2_canopen problem, and I have solved this issue. Thank you!

closed 2024-07-18 by 15906185391