ros-controls/ros2_controllers #1218: Message subscription in tests is flaky

christophfroehlich, 2024-07-18:
From https://github.com/ros-controls/ros2_controllers/actions/runs/9984483861/job/27593769060

```
     /home/runner/work/ros2_controllers/ros2_controllers/.work/target_ws/src/ros2_controllers/ackermann_steering_controller/test/test_ackermann_steering_controller.hpp:232: Failure
    Expected: (max_sub_check_loop_count) >= (0), actual: -1 vs 0
    Test was unable to publish a message through controller/broadcaster update loop
    
    -- run_test.py: return code -11

  <<< error message
```

saikishor, 2024-07-18:
May be we can increase the looping timeouts, as we are only doing it for 5 milliseconds, may be the load of the CPU affects this part

https://github.com/ros-controls/ros2_controllers/blob/96a8d5798c1e2a80b104d2fb3312e1444dc142c5/ackermann_steering_controller/test/test_ackermann_steering_controller.hpp#L219-L225

bmagyar, 2024-07-18:
PR!

saikishor, 2024-07-18:
You have the PR: #1219

closed 2024-07-18 by bmagyar