ros-controls/ros2_controllers #1241: JTC Cancels original and new trajectory when performing trajectory replacement with RT kernel

hunterhuth, 2024-08-05:
**Describe the bug**
Our application builds a trajectory in real time by appending new points to the end of the trajectory. New points are added every 1.5 seconds, and send over the action interface to the joint trajectory controller. When using a standard kernel (not real-time), everything works as expected, a goal cancelled and a goal accepted message is sent to the client every 1.5 seconds. However, with a RT kernel, the JTC will cancel the old trajectory and the new trajectory. This causes the robot to stop moving until it receives a new trajectory 1.5 seconds later.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a rt kernel
2. send a new trajectory every 1.5 seconds

**Expected behavior**
The controller will cancel 1 trajectory and accept 1 new trajectory every time a new trajectory is sent.

**Screenshots**
[image]
Figure 1. Client side output
[image]
Figure 2. Server side ouput

The lines of interest are between times 1722870436.911681030 and 1722870441.413748510.

**Environment (please complete the following information):**
 - OS: 5.15.0-1052-realtime
 - Version: Humble

christophfroehlich, 2024-08-05:
Hi and thanks for reporting the issue. Could you maybe create a minimum example to reproduce your issue? You could choose something from the ros2_control_demos example repository and write a simple program to repeatedly add new waypoints.

hunterhuth, 2024-08-06:
[bug_example.zip](https://github.com/user-attachments/files/16515295/bug_example.zip)
I attached two files. The first is an adaptation of the ros_control_example_1, that uses a joint trajectory controller. The seconds is a ros package that will send trajectory updates containing position and velocity information every 1.5 seconds. The trajectory replacement code can be seen below along with the terminal commands to run the programs. Please let me know if there is more needed.

step 1)  ros2 launch ros2_control_demo_example_1 rrbot.launch.py 
step 2) ros2 run trajectory_replacement trajectory_replacement 

`#include "rclcpp/rclcpp.hpp"
#include "rclcpp/wait_for_message.hpp"
#include "rclcpp/executor.hpp"
#include "rclcpp_action/rclcpp_action.hpp"

#include "std_msgs/msg/string.hpp"
#include "sensor_msgs/msg/joint_state.hpp"
#include <control_msgs/action/follow_joint_trajectory.hpp>
#include <geometry_msgs/msg/pose.hpp>

#include <fstream>
#include <iostream>
#include <string>
#include <chrono>
#include <memory>

using namespace std::chrono_literals;

class TrajectoryReplacement : public rclcpp::Node
{
public:
    using FollowJointTrajectory = control_msgs::action::FollowJointTrajectory;
    using GoalHandleFollowJointTrajectory = rclcpp_action::ClientGoalHandle<FollowJointTrajectory>;

    TrajectoryReplacement()
        : Node("trajectory_replacement")
    {
        using namespace std::placeholders;

        for (int i = 0; i < 2; i++)
        {
            control_msgs::msg::JointTolerance tol;
            tol.position = -1;
            tol.velocity = -1;
            tol.acceleration = -1;
            joint_tolerance.push_back(tol);
        }

        // setup callback functions
        this->client_ptr_ = rclcpp_action::create_client<FollowJointTrajectory>(
            this,
            "/joint_trajectory_controller/follow_joint_trajectory");

        send_goal_options.goal_response_callback =
            std::bind(&TrajectoryReplacement::goal_response_callback, this, _1);

        send_goal_options.feedback_callback =
            std::bind(&TrajectoryReplacement::feedback_callback, this, std::placeholders::_1, std::placeholders::_2);

        send_goal_options.result_callback =
            std::bind(&TrajectoryReplacement::result_callback, this, std::placeholders::_1);

        // setup timer for adding new points to trajectory
        path_timer_ = this->create_wall_timer(
            1.5s, std::bind(&TrajectoryReplacement::update_trajectory, this));

        send_initial_trajectory();
    }

    void send_initial_trajectory()
    {

        current_time = 1.5;
        //float time = current_time;
[... 146 lines cut ...]

            RCLCPP_ERROR(this->get_logger(), "Goal was canceled");
            break;
        default:
            RCLCPP_ERROR(this->get_logger(), "Unknown result code");
            break;
        }
    }
};

int main(int argc, char *argv[])
{
    rclcpp::init(argc, argv);

    rclcpp::executors::MultiThreadedExecutor executor;
    auto trajReplacement = std::make_shared<TrajectoryReplacement>();
    rclcpp::spin(trajReplacement);
    /*executor.add_node(trajReplacement);
    executor.spin_some();

    trajReplacement->send_initial_trajectory();
    executor.spin();*/

    return 0;
}`

christophfroehlich, 2024-08-07:
have you tried to reproduce the issue with a [simple action server](https://github.com/ros2/examples/tree/rolling/rclcpp/actions), without ros2_control being involved? Then we could be sure that your problem is located in ros2_control, or upstream.

hunterhuth, 2024-08-07:
I can do that, but I do not believe it will have the problem because the minimal action server does not preempt the current goal when a new one is received. However, the joint trajectory controller preempts the active goal when it accepts it. It is also worth noting that I have not observed this problem when trajectories are sent with only position values and not velocity values.

christophfroehlich, 2024-08-07:
I see. (And it seems that there is no native preemption support in the design of ROS 2 yet at all.)

This might be an issue from here
https://github.com/ros-controls/ros2_controllers/blob/fa301f43e6b85d364b42247d8f1b6594cc5ce6c1/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L1482-L1493
or the `RealtimeServerGoalHandle` class itself.
Can you run `colcon test` on the realtime_tools? If they succeed, maybe one can adapt the tests [here](https://github.com/ros-controls/realtime_tools/blob/master/test/realtime_server_goal_handle_tests.cpp) to run into that issue. 

The problem is that we can't test the RT kernel within our CI, and I personally don't have access to any machine with rt kernel currently.

hunterhuth, 2024-08-08:
On the RT kernel, all tests passed, so I will look into making a test case for the issue we see on a real-time kernel

hunterhuth, 2024-08-13:
I haven't had a chance to create the test cases yet, but I have a theory about a race condition causing these errors. When preempting an active goal, it calls the setCanceled for the current rt_active_goal_ and writes a new RealtimeGoalHandlePtr() into the buffer. When returning to the goal accepted callback a new RealtimeGoalHandlePtr is created with the new trajectory and written to the buffer. I think the RT update function is reading from the buffer between when the preempt function writes to the buffer and when the goal_accepted_callback writes to the buffer.

This would explain why the server side output when this error occurs shows the "Goal Reached, Success!" and sends the goal canceled to the client twice.

christophfroehlich, 2024-08-13:
This sounds like a possible reason.