ros-controls/ros2_controllers #1243: 6-wheel rover controller

sees1, 2024-08-07:
I realize a six-wheel controller for lunar rover on ros2_controller base now and maybe it relevant for this repo. 
I can modify steering controller library or write controller from scratch.

christophfroehlich, 2024-08-07:
Can you outline the kinematics of such a rover? It isn't a skid steer, you think of a rocker-bogie suspension system?

sees1, 2024-08-08:
yeah, like rocker-bogie with front and rear steering pair of wheel, where  front and rear pair of wheels turn in the same direction.

christophfroehlich, 2024-08-12:
Do you need other kinematic equations like described [here](https://control.ros.org/rolling/doc/ros2_controllers/doc/mobile_robot_kinematics.html)?
If yes, please start with a PR updating the docs.

Then: we could implement this by refactoring the steering_controllers_library by not implementing the resulting vehicle-kinematics in the lib, but only axis-specific kinematics and the composition is done in the respective controller implementations. This would also help here #692. But this is a bigger change as what you are asking for.

silanus23, 2025-07-05:
Hello @christophfroehlich and everyone,

I came across this issue while working on my own project involving a six-wheel rover. If no one is actively working on this, I'd like to contribute, as I'm already working on a similar implementation.

My understanding of the plan is to:

Define and document the kinematic equations for this type of steering system.

Then, refactor the steering_controllers_library to separate vehicle-kinematics from axis-specific kinematics.

Just wanted to confirm, is this still the desired path forward? If so, I'm happy to start drafting the documentation for the kinematics.

Looking forward to your guidance.