ros-controls/ros2_controllers #1245: [PID][pid_controller] "i_clamp" is active even when "antiwindup" is set to false.

ViktorCVS, 2024-08-09:
**Describe the bug**
"i_clamp_max" and "i_clamp_min" is limiting the "i_error" even when "antiwindup" is set to false. The description  of "i_clamp_max" and "i_clamp_min" given the understand that they only limit the "i_error" when "antiwindup" is set to true.

**To Reproduce**
Steps to reproduce the behavior:
1. Set a PID controller with "antiwindup" set to false
2. Set "i_clamp_max" and "i_clamp_min" to 0.0*
3. Set p and i different from 0.0
4. Execute the simulation** with the controller and send a reference
5. See the error - in this case, the 'i' pid parameter doesn't do anything, because "i_clamp_max" and "i_clamp_min" is capping the "i_error" to 0.0. 

* Even when I don't set them, due to the dafault value, they are set to 0.0
** I faced this issue while on simulation

**Expected behavior**
I expected that if "antiwindup" is set to false, "i_clamp_max" and "i_clamp_min" won't cap the "i_error" and the "i" parameter would have effect on the control.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.4 LTS
 - Version: Iron

**Additional context**
In screenshot, is visible that besides the "i_error" is increasing, the output is constant. The reference of velocity, 0.1, never get achieved. In RVIZ, the simulation is coherent with the data. I made new tests, setting "antiwindup" to false and "i_clamp_max" and "i_clamp_min" to a high value, like +-100000 and it worked and "i_error" isn't limited.

christophfroehlich, 2024-08-09:
I had a quick look, it should be considered with the PID ros2_controller. And this uses the pid_ros from control_toolbox package, where we have tests for that.

Only to rule out the simple mistakes: Have you checked if the correct parameter is set? I has to be `gains.<joint_names>.antiwindup` for every joint. (for example, use rqt_dynamic_reconfigure).

ViktorCVS, 2024-08-09:
Thanks for the answer!

Here the full code block from controllers.yaml

`pid_controller_test:
  ros__parameters:
    dof_names:
      - joint1
      - joint2
    command_interface: velocity
    reference_and_state_interfaces:
      - velocity
    gains:
      joint1:
        p: 0.2
        i: 1.0
        d: 0.0
        antiwindup: false
        i_clamp_max: 0.0
        i_clamp_min: 0.0
      joint2:
        p: 0.6
        i: 0.3
        d: 0.0
        antiwindup: false
        i_clamp_max: 0.0
        i_clamp_min: 0.0`

ViktorCVS, 2024-08-09:
Thanks for the answer!

Here the full code block from controllers.yaml

`
pid_controller_test:
  ros__parameters:
    dof_names:
      - joint1
      - joint2
    command_interface: velocity
    reference_and_state_interfaces:
      - velocity
    gains:
      joint1:
        p: 0.2
        i: 1.0
        d: 0.0
        antiwindup: false
        i_clamp_max: 0.0
        i_clamp_min: 0.0
      joint2:
        p: 0.6
        i: 0.3
        d: 0.0
        antiwindup: false
        i_clamp_max: 0.0
        i_clamp_min: 0.0
`

ViktorCVS, 2024-08-09:
Thanks for the answer!

Here the full code block from controllers.yaml

[image]

closed 2025-01-21 by ViktorCVS