ros-controls/ros2_controllers #1246: Problem remapping joint_state_broadcaster

brennand, 2024-08-11:
**Describe the bug**
I have 2 control managers and want to remap one of the joint_state_broadcaster output topic to `/joint_states`.  I've tried a lot of different combinations and still no luck.

This is what my launch file looks like.

```
    gripper_joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="spawner",
        namespace="gripper_ns",
        arguments=[
            "joint_state_broadcaster",
            "--controller-manager",
            "/gripper_ns/controller_manager",
        ],
        remappings=[
            ("/gripper_ns/joint_states", "/joint_states"),
        ],
        output={"both": "screen"},
        emulate_tty=True,
    )
```

**Expected behavior**
I would expect that the `joint_states` topic should change in `ros2 topic list`

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Version: humble

christophfroehlich, 2024-08-12:
You have to add the remapping to the ros2_control_node, because the node od the controller will be created from there. The spawner is only a helper script which interacts with the controller_manager (also launched from ros2_control_node).

brennand, 2024-08-12:
@christophfroehlich , Thank you for your fast reply.

closed 2024-08-12 by brennand