ros-controls/ros2_controllers #1263: Chainable controllers

Richard-Haes-Ellis, 2024-08-22:
I am often frustrated by the current limitations in ROS 2 Control when it comes to chaining controllers. Specifically, the inability to directly connect higher-level controllers, such as custom Linear Quadratic Regulators (LQR) or Model Predictive Controllers (MPC), with lower-level controllers (like position, velocity, or effort controllers) restricts the flexibility and sophistication of control strategies in complex robotic systems. This limitation makes it challenging to implement advanced control strategies where higher-level controllers need to dictate the references or setpoints for lower-level controllers in real-time.

I propose adding support for chaining controllers within the ROS 2 Control framework. The ability to configure higher-level custom controllers (such as LQR, MPC) to directly control the references or setpoints of lower-level controllers (such as position, velocity, and effort controllers) would significantly enhance the framework's capabilities. This feature would make ROS 2 Control more powerful and versatile, comparable to advanced systems like Simulink.

Here’s what I envision:

- Controller Composition: Enable configuration of higher-level controllers to send commands or setpoints to lower-level controllers through a well-defined interface or communication channel.

- Chainable Controllers: Enhance the existing ControllerInterface class to support chaining, allowing for integration of existing controllers in a hierarchical manner.

Alternatives:

While there have been some attempts to demonstrate controller chaining (e.g., [chaining_demo](https://github.com/ChiratheRobotics/chaining_demo)), these are not yet integrated into the mainline ROS 2 Control framework. Documentation on controller chaining ([controller chaining](https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/controller_chaining.html)) provides some insights but lacks practical implementation.

Currently, advanced control algorithms (AFAIK) must be implemented in separate nodes that subscribe to and publish controller inputs and outputs. This approach compromises real-time performance and creates complexity, making the system less clean and manageable. Integrating chaining directly into the ROS 2 Control framework would offer a more streamlined and real-time-capable solution.

Additional context

Integrating controller chaining within ROS 2 Control would align with the broader goal of enhancing ROS 2's capabilities for advanced robotics applications. It would facilitate more sophisticated control strategies and increase ROS 2 Control’s competitiveness with other advanced control frameworks.

Richard-Haes-Ellis, 2024-08-22:
It seems that there is implementation of this feature in the ControllerInterface class but not sure how it's used in the controllers. 
https://github.com/ros-controls/ros2_control/blob/master/controller_interface/src/controller_interface.cpp

Richard-Haes-Ellis, 2024-08-22:
Looks like there has been efforts on the diff_drive controller: https://github.com/ros-controls/ros2_controllers/commit/dea3d3c027ef051a6a159264749bacab7065ecee#diff-f8d3a930a6782d85b295ef6bce89dc2a757ae540cc80128e8044aee6ddee5df9

christophfroehlich, 2024-08-22:
Fyi there is a chaining demo in our [examples repository](https://control.ros.org/rolling/doc/ros2_control_demos/example_12/doc/userdoc.html).

I'm not sure why your LQR or MPC cannot directly claim the interfaces. If it is just a naming issue (like the MPC has `controller_output` which you want to connect to `joint1/velocity`), then https://github.com/ros-controls/ros2_control/pull/1667 will help you.