ros-controls/ros2_controllers #1283: Ackermann: Steering angle is depending on robot speed

LukasLange28, 2024-09-03:
I am currently working on a four wheel robot (One axis for steering and one axis for driving).
I have implemented the ackermann controller for the robot.

The ackermann controller does not give out an steering angle, when the speed (linear x value) is zero.
Is there an option to make this possible or does the ackermann controller not supported steering in still stand?

The calculated steering angle is also depending on the speed of the robot.
Is there an option to make the steering angle independent of the robot speed?

christophfroehlich, 2024-09-05:
> I am currently working on a four wheel robot (One axis for steering and one axis for driving). I have implemented the ackermann controller for the robot.
>
> The ackermann controller does not give out [...]

If you have steering on a different axis than traction, how could you move in still stand?

> The calculated steering angle is also depending on the speed of the robot. Is there an option to make the steering angle independent of the robot speed?

Then you don't need the ackermann_steering_controller: It takes a twist as input and applies the vehicle kinematics to calculate steering and traction commands. If you want to set that independently, just use a feedforward controller.

LukasLange28, 2024-09-09:
> If you have steering on a different axis than traction, how could you move in still stand?

When it the robot is doing stuff like parking, it would be helpful, to steer the robot before moving.

> Then you don't need the ackermann_steering_controller: It takes a twist as input and applies the vehicle kinematics to calculate steering and traction commands. If you want to set that independently, just use a feedforward controller.

Thanks for advice. I am going to try this.

closed 2024-09-09 by LukasLange28

christophfroehlich, 2024-09-09:
the ackermann kinematics assume that there are no velocity limits on the steering axis, i.e., the desired steering angle is immediately set. Would it be helpful to limit the traction output if the steering angle is not reached yet? This is implemented in the tricycle_controller, but not yet in the steering_controllers_library
https://github.com/ros-controls/ros2_controllers/blob/c08bdab049cc090df0643d3399bf53862f4571be/tricycle_controller/src/tricycle_controller.cpp#L173-L189