ros-controls/ros2_controllers #1292: Issues building inside IsaacROSDev containers

brettpac, 2024-09-22:
I'm trying to take the first step of building ros2_controllers inside of the IsaacROSDev containers. But, its not compiling due to problems with gmock.

To recreate, follow the readme instructions here...
https://github.com/robosoft-ai/nova_carter_sm_library/tree/main/sm_nav2_test_3

**Environment (please complete the following information):**
 - OS: Host is Ubuntu 22.04
 - Version: Humble. Including the Humble branch of ros2_controllers.

Here's the terminal output...
led,automatic]
```
admin@system76-pc:/workspaces/isaac_ros-dev$ colcon build --symlink-install
[0.512s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
        'isaac_ros_common' is in: /opt/ros/humble
        'isaac_ros_nitros_compressed_image_type' is in: /opt/ros/humble
        'gxf_isaac_optimizer' is in: /opt/ros/humble
        'isaac_ros_managed_nitros' is in: /opt/ros/humble
        'isaac_ros_tensor_rt' is in: /opt/ros/humble
        'isaac_ros_nitros' is in: /opt/ros/humble
        'position_controllers' is in: /opt/ros/humble
        'gxf_isaac_tensorops' is in: /opt/ros/humble
        'isaac_ros_nitros_image_type' is in: /opt/ros/humble
        'isaac_ros_dnn_image_encoder' is in: /opt/ros/humble
        'isaac_ros_image_proc' is in: /opt/ros/humble
        'nav2_map_server' is in: /opt/ros/humble
        'isaac_ros_launch_utils' is in: /opt/ros/humble
        'gripper_controllers' is in: /opt/ros/humble
        'isaac_ros_test' is in: /opt/ros/humble
        'isaac_ros_gxf' is in: /opt/ros/humble
        'isaac_ros_tensor_list_interfaces' is in: /opt/ros/humble
        'isaac_ros_nitros_detection2_d_array_type' is in: /opt/ros/humble
        'owl_description' is in: /opt/ros/humble
        'isaac_ros_triton' is in: /opt/ros/humble
        'joint_trajectory_controller' is in: /opt/ros/humble
        'hawk_description' is in: /opt/ros/humble
        'joint_state_broadcaster' is in: /opt/ros/humble
        'isaac_ros_nitros_tensor_list_type' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
        --allow-overriding gripper_controllers gxf_isaac_optimizer gxf_isaac_tensorops hawk_description isaac_ros_common isaac_ros_dnn_image_encoder isaac_ros_gxf isaac_ros_image_proc isaac_ros_launch_utils isaac_ros_managed_nitros isaac_ros_nitros isaac_ros_nitros_compressed_image_type isaac_ros_nitros_detection2_d_array_type isaac_ros_nitros_image_type isaac_ros_nitros_tensor_list_type isaac_ros_tensor_list_interfaces isaac_ros_tensor_rt isaac_ros_test isaac_ros_triton joint_state_broadcaster joint_trajectory_controller nav2_map_server owl_description position_controllers

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> isaac_ros_common
Starting >>> magic_enum
Starting >>> isaac_ros_test
Starting >>> nav2_common
Starting >>> smacc2_msgs
Starting >>> isaac_ros_tensor_list_interfaces
Starting >>> nav_2d_msgs
Starting >>> isaac_ros_apriltag_interfaces
Starting >>> isaac_ros_nova_interfaces
Starting >>> isaac_ros_launch_utils
Starting >>> nav2z_planners_common
Starting >>> isaac_ros_pointcloud_interfaces
Starting >>> isaac_ros_json_info_generator
Starting >>> nvblox_msgs
Starting >>> vda5050_msgs
Starting >>> nvblox_ros_common
Starting >>> nvblox_ros_python_utils
Starting >>> forward_command_controller
Starting >>> steering_controllers_library
Starting >>> joint_trajectory_controller
Starting >>> sllidar_ros2
Starting >>> isaac_ros_bi3d_interfaces
Starting >>> isaac_ros_perceptor_constants
Starting >>> joint_state_broadcaster
Starting >>> diff_drive_controller
[... 634 lines cut ...]
ros-humble-twist-mux-msgs/jammy,now 3.0.1-1jammy.20240728.211903 amd64 [installed,automatic]
ros-humble-twist-mux/jammy,now 4.3.0-1jammy.20240728.225624 amd64 [installed]
ros-humble-uncrustify-vendor/jammy,now 2.0.2-2jammy.20240728.194050 amd64 [installed,automatic]
ros-humble-unique-identifier-msgs/jammy,now 2.2.1-3jammy.20240728.211151 amd64 [installed,automatic]
ros-humble-ur-description/jammy,now 2.1.6-1jammy.20240822.174237 amd64 [installed]
ros-humble-ur-moveit-config/jammy,now 2.2.15-1jammy.20240823.154355 amd64 [installed]
ros-humble-ur-msgs/jammy,now 2.0.0-2jammy.20240728.220903 amd64 [installed]
ros-humble-urdf-parser-plugin/jammy,now 2.6.1-1jammy.20240822.151133 amd64 [installed,automatic]
ros-humble-urdf/jammy,now 2.6.1-1jammy.20240822.151433 amd64 [installed,automatic]
ros-humble-urdfdom-headers/jammy,now 1.0.6-2jammy.20240728.192446 amd64 [installed,automatic]
ros-humble-urdfdom-py/jammy,now 1.2.1-1jammy.20240728.222109 amd64 [installed,automatic]
ros-humble-urdfdom/jammy,now 3.0.2-2jammy.20240728.202538 amd64 [installed,automatic]
ros-humble-v4l2-camera/jammy,now 0.6.2-1jammy.20240820.161745 amd64 [installed]
ros-humble-vda5050-msgs/jammy,now 3.0.1-0jammy amd64 [installed,automatic]
ros-humble-vision-msgs-rviz-plugins/jammy,now 4.1.1-1jammy.20240822.174109 amd64 [installed]
ros-humble-vision-msgs/jammy,now 4.1.1-1jammy.20240728.220906 amd64 [installed]
ros-humble-vision-opencv/jammy,now 3.2.1-1jammy.20240728.222948 amd64 [installed]
ros-humble-visualization-msgs/jammy,now 4.2.4-1jammy.20240728.221458 amd64 [installed,automatic]
ros-humble-warehouse-ros-sqlite/jammy,now 1.0.4-1jammy.20240820.171150 amd64 [installed,automatic]
ros-humble-warehouse-ros/jammy,now 2.0.4-3jammy.20240820.165308 amd64 [installed,automatic]
ros-humble-xacro/jammy,now 2.0.8-1jammy.20240728.200111 amd64 [installed]
ros-humble-yaml-cpp-vendor/jammy,now 8.0.2-1jammy.20240728.194132 amd64 [installed,automatic]
ros-humble-zstd-vendor/jammy,now 0.15.12-1jammy.20240730.141109 amd64 [installed,automatic]
admin@system76-pc:/workspaces/isaac_ros-dev$ 
```

christophfroehlich, 2024-09-23:
Maybe someone finds time trying to reproduce this in your container. In the meantime, could you try to compile with `-DBUILD_TESTING=OFF`, e.g., by using the mixin
https://github.com/colcon/colcon-mixin-repository/blob/master/build-testing.mixin

brettpac, 2024-09-23:
It worked! Thank you @christophfroehlich
Final command that worked was 
`colcon build --symlink-install --cmake-args -DBUILD_TESTING=OFF`

leonce-m, 2025-04-07:
This is an unacceptable workaround. We have the same issue and specifically need to run our unit tests in the Isaac ROS docker container!

christophfroehlich, 2025-04-07:
Then you have to consult the author of your docker container.. 
Our setup works on several systems in our CI and on the ROS buildfarm without issues.

leonce-m, 2025-04-07:
IRDC is a giant 37.2GB big pile of trash that we unfortunately can't get rid of at the moment and I don't expect Nvidia to do anything about this as their usual response in their forums boils down to "skill issue; fix it yourself" - which is what I'm trying to do. But I just can't wrap my head around why ament fails to find gmock_vendor and gtest_vendor. It doesn't make any sense.

leonce-m, 2025-04-07:
I have added
```
RUN --mount=type=cache,target=/var/cache/apt \
    apt-get update && apt-get install --reinstall --no-install-recommends -y \
    ros-humble-gmock-vendor \
    ros-humble-gtest-vendor
```
to our final build layer and it seems to have solved the issue at the moment. This was necessary even though according to dpkg and apt they were already installed.

saikishor, 2025-04-08:
@brettpac @leonce-m we don't have any near future plans to add support for IssacROSDevContainer. We have CI running for the Tier-1 supported systems, until here we can ensure that it is working well. On the other hand, like any other open-source projects we also accept contributions. If anyone has a fix for this problem, please open a PR, we will be happy to review them and get them merged. 

Have a great day 😀

christophfroehlich, 2025-04-08:
Not only Tier 1 but RHEL debian and partially even Windows ;)
I help where possible, but in this case it wasn't.

@leonce-m has already posted a fix, I don't think that we can do more about that. If you guys think we should add these step to the installation instructions to a more prominent position, we are happy for any contributions on control.ros.org

closed 2025-04-08 by christophfroehlich

bmagyar, 2025-04-08:
Aye, I agree with @leonce-m about Nvidia software in general being a pile of trash. If that container doesn't have a usable installation of dependencies, there is only so much we can do. I quite like the extra layer on top, fixing the packages.

@christophfroehlich I was about to suggest the exact same thing. We should perhaps add a new section on "installation" with a specific subsection for the isaac ros dev containers and we can collect fixes for all problem children there.

brettpac, 2025-04-08:
@bmagyar, @christophfroehlich I don't think the invectives being used here regarding IRDC are accurate or helpful.

I mean, I've recently and personally used IRDCs to demo the NVIDIA IsaacROS Vision Pipeline, and found the IRDCs (which are really a sound containerization strategy) to be a game changer because of the ability to get assistance on the NVIDIA Forums.

And IRDCs may not be perfect yet, but if IRDCs are trash, what are the ROS2 rmw layer and Gazebo? Solid gold? please.

Is the Jetson platform trash too? In the next 5 years, do you think more people are going to attempt to deploy ROS2 (and maybe even ros2_control) onto RHEL and Windows devices, or the Jetson platform? And if they use RHEL and Windows, what vision pipeline are they going to use?

IRDCs are working fine, and are able to handle their dependencies. The issue here is a narrow one related to the uncommon use of gmock by this repo, which is fine, but that's what it is.