ros-controls/ros2_controllers #1298: joint_trajectory_controller interpolation change between Humble and Jazzy

mrjogo, 2024-09-26:
**Describe the bug**
I have just upgraded from Humble to Jazzy, and a trajectory that previously executed correctly is now deviating quite a bit, which my guess is from the controller interpolation.

I'm using `moveit_task_constructor` to create a cartesian UR5e trajectory that contains a segment with a very sharp jerk (essentially three co-linear waypoints, each in an opposite direction). When I visualize / plot the planned trajectory in rviz, it is correct:

[image]

However, when I execute the task, the shake gets wildly exaggerated (this behavior also occurs on a real UR5e):

[image]

Plotting the `shoulder_pan_joint` positions from  `/joint_trajectory_controller/follow_joint_trajectory/_action/feedback` shows that the motion matches "desired" (in the planned trajectory, `shoulder_pan_joint` is supposed to go to 0.996, not all the way to -0.76):

[image]

I'm not super familiar with the internals of `joint_trajectory_controller`, but did something change since Humble? Is this a bug, or expected interpolation behavior given the jerkiness of the commanded trajectory? If the latter, what's the right way to command such motions?

**To Reproduce**
Steps to reproduce the behavior:

Execute the following RobotTrajectory:

```yaml
# moveit_msgs/RobotTrajectory.msg
joint_trajectory:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: "world"
  joint_names:
  - "shoulder_pan_joint"
  - "shoulder_lift_joint"
  - "elbow_joint"
  - "wrist_1_joint"
  - "wrist_2_joint"
  - "wrist_3_joint"
  points:
  -
    positions:
    - 1.24934
    - -1.54731
    - 1.91419
    - -0.368635
    - 1.25010
    - -1.56987
    velocities:
    - -0.00398828
    - -2.05360e-06
    - 0.000135010
    - -0.000151254
    - -0.00398789
    - -0.00800000
    accelerations:
    - -3.98828
    - -0.00205360
    - 0.135010
    - -0.151254
    - -3.98789
    - -8.00000
    effort: []
    time_from_start:
      sec: 0
      nanosec: 0
  -
    positions:
    - 1.22943
    - -1.54725
    - 1.91480
    - -0.369390
[... 799 lines cut ...]
      nanosec: 540359020
multi_dof_joint_trajectory:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ""
  joint_names: []
  points: []
```

**Expected behavior**
A short shaking motion.

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - ROS: Jazzy
 - joint_trajectory_controller: 4.14.0
- ur_robot_driver: 2.4.10
- moveit_task_constructor: https://github.com/moveit/moveit_task_constructor/tree/cb867aeddfdca6e3791a4ede998fe23ce9958754

**Additional context**
The controller configuration used is https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/2.4.10/ur_robot_driver/config/ur_controllers.yaml, which doesn't seem to have changed materially since Humble.

The commanded trajectory comes from the result of the MTC `Task::plan` call, which is then executed with `Task::execute`. I'm making the assumption that MTC doesn't alter the trajectory in `execute` before it sends it to the controller.

christophfroehlich, 2024-09-27:
Release notes are summarized in https://control.ros.org/jazzy/doc/ros2_controllers/doc/release_notes.html#joint-trajectory-controller but I can't recall any changes which could explain your behavior.

Have you tried to replay the posted trajectory in your humble setup, just to eliminate the effects of any possible changes from MoveIt?

mrjogo, 2024-09-28:
Good suggestion.

TL;DR A change in `trajectory_processing::TimeOptimalTrajectoryGeneration` between Humble and Jazzy fully explains the observed motion. Thanks for your time; I'll close this issue.

Like you suggested, I created a little script (pasted down below for posterity) to directly call `FollowJointTrajectory`, and it showed that for a given trajectory, the observed motion was the same between Humble and Jazzy.

Given that, I compared the correctly behaving trajectory that Humble generated with Jazzy. The difference turns out to be that in Humble, `trajectory_processing::TimeOptimalTrajectoryGeneration` gives the first point 0 velocity and acceleration, but in Jazzy it gives it nonzero velocity and acceleration. Because my code appends two trajectories with a long (~3.5s) pause in between (during which time the robot shouldn't move), with the Jazzy trajectory, the JointTrajectoryController receives sequential points that are 3.5 seconds apart with identical positions, but the first has 0 velocity / acceleration and the second does not. The interpolator must then create the large motion to best achieve both those states. It turns out the robot was moving from the "pause", and the jerky shake was a red herring.

Here's my `test_joint_trajectory.py` script. The `joint_trajectory.yaml` file is, you guessed it, a `trajectory_msgs/JointTrajectory` message saved as YAML.

```python
#!/usr/bin/env python3

import sys

import rclpy
import yaml
from control_msgs.action import FollowJointTrajectory
from moveit_msgs.action import MoveGroup
from moveit_msgs.msg import Constraints, JointConstraint
from rclpy.action import ActionClient
from rclpy.node import Node
from rosidl_runtime_py import message_to_yaml, set_message_fields
from trajectory_msgs.msg import JointTrajectory

class FollowJointTrajectoryActionClient(Node):

    def __init__(self, joint_trajectory_yaml, manipulator_group_name="manipulator"):
        super().__init__("follow_joint_trajectory_action_client")
        self._manipulator_group_name = manipulator_group_name

        self._follow_joint_trajectory_client = ActionClient(
            self,
            FollowJointTrajectory,
            "/joint_trajectory_controller/follow_joint_trajectory",
        )
        self._move_group_action_client = ActionClient(self, MoveGroup, "/move_action")
        self._joint_trajectory_msg = JointTrajectory()
        joint_trajectory_dict = yaml.safe_load(joint_trajectory_yaml)
        set_message_fields(self._joint_trajectory_msg, joint_trajectory_dict)

        self._executed_trajectory = []

    def move_to_start(self):
        self.get_logger().info("Moving to start position")
        goal_msg = MoveGroup.Goal()
        goal_msg.request.group_name = self._manipulator_group_name
        joint_goal = Constraints()
        for joint_name, position in zip(
            self._joint_trajectory_msg.joint_names,
            self._joint_trajectory_msg.points[0].positions,
        ):
            joint_constraint = JointConstraint()
            joint_constraint.joint_name = joint_name
            joint_constraint.position = position
            joint_constraint.tolerance_above = 0.002
            joint_constraint.tolerance_below = 0.002
            joint_constraint.weight = 1.0
            joint_goal.joint_constraints.append(joint_constraint)

        goal_msg.request.goal_constraints.append(joint_goal)

        self._move_group_action_client.wait_for_server()

        move_to_start_future = self._move_group_action_client.send_goal_async(goal_msg)
        move_to_start_future.add_done_callback(self._move_to_start_response_callback)

    def _move_to_start_response_callback(self, future):
        goal_handle = future.result()
        if not goal_handle.accepted:
[... 59 lines cut ...]
    rclpy.init(args=args)

    if len(sys.argv) != 3:
        print(
            "Usage: test_joint_trajectory.py <manipulator_group_name> <path/to/joint_trajectory.yaml>"
        )
        return

    manipulator_group_name = sys.argv[1]
    joint_trajectory_file_path = sys.argv[2]

    with open(joint_trajectory_file_path, "r") as file:
        joint_trajectory_yaml = file.read()

    action_client = FollowJointTrajectoryActionClient(
        joint_trajectory_yaml, manipulator_group_name=manipulator_group_name
    )
    action_client.move_to_start()
    rclpy.spin(action_client)

    action_client.save_feedback_yaml()

if __name__ == "__main__":
    main()
```

closed 2024-09-28 by mrjogo

saikishor, 2024-09-28:
Hello @mrjogo!

Thanks for the script. As you found that the issue is in `trajectory_processing::TimeOptimalTrajectoryGeneration`, if you are opening a corresponding issue in MoveIt2, please leave a reference to that issue here.

Thank you!