ros-controls/ros2_controllers #1312: Ask about chaining YAML configs

drfenixion, 2024-10-10:
I am integrating ros2_controllers to RobotCAD and faced to some lack of examples of chaining ros2_controllers.

I found some examples [https://github.com/ros-controls/control.ros.org/blob/31f2c1d9c30aa0d4853d8d9e3c18b7516013dde5/doc/resources/urdfs%20and%20yamls/example_controllers.yaml#L123](https://github.com/ros-controls/control.ros.org/blob/31f2c1d9c30aa0d4853d8d9e3c18b7516013dde5/doc/resources/urdfs%20and%20yamls/example_controllers.yaml#L123)
But it is not clear for me and looks like outdate or future because of i dont see limiting_controllers/JointLimitingController in ros2_controllers. F.e. how can i command to PID controller from any other controller? I about YAML config options.

In **PID_controller** i see 
https://github.com/ros-controls/ros2_controllers/blob/97c1e2472f2bbc98a9cb86449b41611a3e298b30/pid_controller/src/pid_controller.cpp#L392
And looks like it is take reference from reference_and_state_dof_names but what i should fill here? Is it something like this: `precedent_controller_name/joint` or just `joint`? 
What if i want to fill YAML config of precedent controller instead of PID controller? Like `command_joints` option of admittance_controller and joint_trajectory_controller.

**Steering library**
https://github.com/ros-controls/ros2_controllers/blob/97c1e2472f2bbc98a9cb86449b41611a3e298b30/steering_controllers_library/src/steering_controllers_library.cpp#L318
There is present `reference_interfaces` but it is not present in [https://github.com/ros-controls/ros2_controllers/blob/master/steering_controllers_library/src/steering_controllers_library.yaml](https://github.com/ros-controls/ros2_controllers/blob/master/steering_controllers_library/src/steering_controllers_library.yaml). How i should use it in YAML config of controller?

**Please clearify chaining config params for me or better give examples of chaining of all chanable ros2_controllers.**

Btw, at screenshoot my current progress of integration of ros2_controllers to RobotCAD. This will be available in one of RobotCAD future versions. 
[image]

christophfroehlich, 2024-10-17:
Your linked yaml file is rather old tbh.
There is a [workshop page](https://github.com/ros-controls/roscon2023_control_workshop) from last year's ROSCon, there will be another workshop next Monday, [repo is still WIP](https://github.com/ros-controls/roscon2024_control_workshop) I guess.

drfenixion, 2024-10-18:
@christophfroehlich, thank you for links, there are lot of useful but only 2 examples of chaining currently.

[https://github.com/ros-controls/roscon2023_control_workshop/blob/rolling/tiago_chaining/config/controllers.yaml#L55](https://github.com/ros-controls/roscon2023_control_workshop/blob/rolling/tiago_chaining/config/controllers.yaml#L55)

[https://github.com/ros-controls/roscon2024_control_workshop/blob/a312d3180c2da0edb7768e0fda04e2ef498df0b0/workshop_bringup/config/chaining_controllers.yaml#L76](https://github.com/ros-controls/roscon2024_control_workshop/blob/a312d3180c2da0edb7768e0fda04e2ef498df0b0/workshop_bringup/config/chaining_controllers.yaml#L76)

And both examples with castom controllers.

I need examples with all standart ros2_controllers because I need fully understand how to generate their configs from RobotCAD GUI.

Also it nice to have option in controllers params like - `mandatory: true` in future. Because i am using sophisticated logic to filter mandatory params for GUI.
Example of config file where could be mandatory option of every param.
[https://github.com/ros-controls/ros2_controllers/blob/master/ackermann_steering_controller/src/ackermann_steering_controller.yaml](https://github.com/ros-controls/ros2_controllers/blob/master/ackermann_steering_controller/src/ackermann_steering_controller.yaml)

christophfroehlich, 2024-10-18:
Not all controllers are chainable, i.e. preceded by other ones, only these here
https://github.com/search?q=repo%3Aros-controls%2Fros2_controllers+chainable_controller_interface.hpp&type=code
others can be after chainable_controllers but not before

drfenixion, 2024-10-18:
@christophfroehlich, I present some examples please correct me if i wrong.
```
admittance_controller
      command_joints:
        - next_controller/joint1
```
        
```
joint_trajectory_controller
      command_joints:
        - admittance_controller/joint1
```
JTC does not have chainable interface but has command_joints params. How it is posible? Just for future implementation but param not used now?
[https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller_parameters.yaml#L11](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller_parameters.yaml#L11)

```
pid_controller
       reference_and_state_dof_names:
             - **HOW to use?** 
```
Looks like it (`reference_and_state_dof_names`) is not for command next controller but for getting command from previous. I dont see option for command next controller like` command_joints`. How to command from PID controller next controller?
[https://github.com/ros-controls/ros2_controllers/blob/master/pid_controller/src/pid_controller.yaml#L12](https://github.com/ros-controls/ros2_controllers/blob/master/pid_controller/src/pid_controller.yaml#L12)

```
pid_controller
       dof_names:
             - next_controller/join1   - **Should it be like this?**
```

[tricycle_steering_controller](url)
How command from it to next controller? Where is no params like `command_joints`.
```
     rear_wheels_names
          - pid_controller/join1 - **Should it be like this?**
     front_wheels_names
         - pid_controller/join1  - **Should it be like this?**
```

I want to derive a general rule for chains. In previous chainable controller i should use command param (`command_joints`) or joints params (dof_names, rear_wheels_names, front_wheels_names) if command param does not present and fill it with string like: `next_controller/join1`. Is it correct?

drfenixion, 2024-11-01:
Is there are send `bicycle steering controller/linear` as reference in chain mode from any other standard controller? I mean via config. 
Or is it possible only from custom controller?

F.e. is it posible something like this?
joint_group_velocity_controller
    joints:
        bicycle steering controller/linear
        bicycle steering controller/angular

christophfroehlich, 2025-03-26:
> Is it possible to send `bicycle steering controller/linear` as reference in chain mode from any other standard controller? I mean via config. Or is it possible only from custom controller?
> 
> F.e. is it posible something like this?
> 
> ```
> joint_group_velocity_controller
>     joints:
>         bicycle_steering_controller/linear
>         bicycle_steering_controller/angular
> ```

exactly, this should be possible.