ros-controls/ros2_controllers #1316: Cleanse the joint_states from old joints (when changing tools)

LucasLabarussiat, 2024-10-14:
Hello,

I am working with an UR10e and different tools, among which a RobotiQ 2F-140 and a RobotiQ E-Pick.
I have a routine to change tools:
- deactivate old tool controllers, then hardware component, close the tool communication and power off the tool
- store the old tool and pick the new one
- power on the new tool, activate the tool communication, then the hardware component and finally the new controller(s)

When picking a new tool, I also deactivate the `joint_state_broadcaster` and then re-activate it : this way, in the `/joint_states` topic, the states of the new tool get added to the ones from the robot.
Unfortunately, the same process does not get rid of the joint(s) from a previous tool. Even if I reboot the `joint_state_broadcaster`, I will still have, for example, the `finger_joint` from the `robotiq_gripper_controller`. The value of course is not updated anymore because the controller has been deactivated, but the fact that it remains in the list of joints bothers me for some applications.

Any idea how I could cleanse the joint_states ? Is it due to the QoS profile of the joint_state_broadcaster ? Here is what I get (I guess it is the default QoS for it since I did not change it):
```
Node name: joint_state_broadcaster
Node namespace: /
Topic type: sensor_msgs/msg/JointState
Endpoint type: PUBLISHER
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```
Or, given that I already deactivated - reactivated the publisher on the topic, could the "remnant" be caused by the subscribers which now "expect" this joint to be there ? I do not know the precise underlying process.

I am not sure I gave sufficient details, please do not hesitate to tell me if that is the case.

Thanks,
Lucas

christophfroehlich, 2024-10-14:
I'm not exactly sure if those are kept in the ressource manager. Do you still see the hardware interfaces with `ros2 control list_hardware_interface` after cleanup?

I just tried example_6 of the demos, and performed a cleanup of one hardware component. The cm still reports them, but as unavailable
```
$ ros2 control list_hardware_interfaces
command interfaces
        joint1/position [available] [unclaimed]
        joint2/position [unavailable] [unclaimed]
state interfaces
        joint1/position
        joint2/position
```
and I still see both joints on /joint_states topic. After unspawning (back to unloaded state) and spawning the broadcaster again -> /joint_states holds only joint1

LucasLabarussiat, 2024-10-15:
Thanks for your quick answer.

> Do you still see the hardware interfaces with `ros2 control list_hardware_interface`
I do yes. In fact, I launch my whole setup with the hardware interfaces of the tools only loaded, with the following config file:
```
controller_manager:
  ros__parameters:
    hardware_components_initial_state:
      unconfigured:
        - "RobotiqGripperHardwareInterface"
        - "EpickGripperHardwareInterface"
```

 I have the same result whether it is before activating a tool or after cleanup on the hardware interfaces side:
```
ros2 control list_hardware_interfaces
command interfaces
	[...]
	finger_joint/position [unavailable] [unclaimed]
	finger_joint/set_gripper_max_effort [unavailable] [unclaimed]
	finger_joint/set_gripper_max_velocity [unavailable] [unclaimed]
	[...]
	vacuum/vacuum_cmd [unavailable] [unclaimed]
	[...]
state interfaces
	[...]
```

> After unspawning (back to unloaded state) and spawning the broadcaster again
Ok this worked for me as well. So when registering a new joint, deactivating and activating back the broadcaster is enough, but to cleanse it, you have to *unspawn* it. Not sure if this is intended ? Anyway this will work with a couple more calls to services. Thanks, I should have tried it myself. Should I close the ticket then ?

On a side note, I also just realized the command line `ros2 lifecycle set` does not work for controllers: `Node not found`, I guess only the `controller_manager` has a say on their state.

closed 2024-10-15 by christophfroehlich