ros-controls/ros2_controllers #1317: About min_acceleration when reversing

otamachan, 2024-10-22:
**Is your feature request related to a problem? Please describe.**
I am using the `diff_drive_controller`, and I want the robot to accelerate slowly when speeding up and decelerate quickly to come to a stop. To achieve this, I have set `min_acceleration` and `max_acceleration`, and the robot behaves as expected when moving forward. However, when reversing, the acceleration is fast, and the deceleration is slow, so it does not work as expected.

**Describe the solution you'd like**
Would it be better to reverse the handling of `min_acceleration` and `max_acceleration` when reversing?

**Describe alternatives you've considered**

**Additional context**
I found the similar issue in ros_controllers(https://github.com/ros-controls/ros_controllers/issues/290)

otamachan, 2024-10-22:
might be related to https://github.com/ros-controls/control_toolbox/pull/212

christophfroehlich, 2024-10-24:
Thanks for reporting this, I'll have a look and fix that maybe with (or after) the PR in control_toolbox. We have a newer implementation of the speed limiter in the tricycle_controller package, where there is also a min_deceleration and max_deceleration parameter. Maybe we should consider using that one for all controllers, but it has different parameters and I'm not sure if we should change the well-established ones from diff_drive_controller.

christophfroehlich, 2024-12-01:
@otamachan could you have a look at #1315 please and see if it solves your problem?

otamachan, 2024-12-03:
Great!! I'll try. Thank you!

PR #1315 (2024-12-16): Update command limiter of diff_drive_controller
https://github.com/ros-controls/control_toolbox/pull/212

It additionally adds parameters ``max_acceleration_reverse`` and ``max_deceleration_reverse`` to configure asymmetric acceleration/deceleration behavior.

Closes #1317
closed 2024-12-16 by christophfroehlich