ros-controls/ros2_controllers #1328: No feedback topic with ackermann controller

LukasLange28, 2024-10-30:
Im working on a carlike robot with an ackermann controller.

The controller works and the robot can already drive.
Now I want to implement a localization for the robot.
Therefor I need the wheel odometry.
As far as I understand it, the ackermann controller can publish a odometry based on the steering joint position.
I already have a node that publishes these positions.

In my yaml file, I have set the "position_feedback" to true.
But I have not found a new topic for these feedback.
What should the feedback topic look like?

Does the hardware interface has an affect on the feedback?
Im working with the "mock_components/GenericSystem".
That worked fine for now.
Do I have to use a different one?
And do I have to write it by my self?
Or can the [carlike hardware interface](https://github.com/ros-controls/ros2_control_demos/blob/master/example_11/hardware/carlikebot_system.cpp) be adapted?

christophfroehlich, 2024-11-02:
"Feedback" here means that measurements from odometry sensors of the wheels are used to calculate the odometry (aka forward kinematics) instead of using the commands sent to the robot. No additional topic will be published, all publishers are summarized [here](https://control.ros.org/rolling/doc/ros2_controllers/steering_controllers_library/doc/userdoc.html#publishers). You maybe look for the `odometry` topic?

LukasLange28, 2024-11-04:
I have a node for each steering wheel, that publishes the measured steering position of each wheel.
That position comes from an angle sensor.
Is there no topic to publish these measured positions?
If not, what is the interface for the sensor data?
Yes, I want to publish an odometry topic, but there for I have to get my "Feedback" into the ackermann controller.

christophfroehlich, 2024-12-17:
Sorry, I have problems to understand your question.
The measurements from your position encoder have to be exported from a hardware component to `<front_wheels_names[i]>/position` in case of front steering, see our [docs](https://control.ros.org/rolling/doc/ros2_controllers/steering_controllers_library/doc/userdoc.html#state-interfaces).
A guideline for writing your hardware component [can be found here](https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/writing_new_hardware_component.html).

LukasLange28, 2024-12-18:
You answered my question.
A hardware component is needed to handle the encoder measurements.

closed 2024-12-18 by LukasLange28