ros-controls/ros2_controllers #1379: Admittance Controller: parameter robot_description not set

mhubii, 2024-11-20:
Hi All,

trying to run the admittance controller. (`Ubuntu 24.04`, `rolling`). Receiving:

```shell
[ros2_control_node-3] [ERROR] [1732101243.342290020] [kinematics_interface_kdl]: parameter robot_description not set
```

Triggered here:
https://github.com/ros-controls/kinematics_interface/blob/822cff404424004d646d384c8dfedb226de2b54a/kinematics_interface_kdl/src/kinematics_interface_kdl.cpp#L42

Configured here:
https://github.com/ros-controls/ros2_controllers/blob/79358e38100139cd1d91ba8d090a388677d464b3/admittance_controller/src/admittance_controller.cpp#L284

The `controller_manager` is name-spaced. Any idea? Thank you!

christophfroehlich, 2024-11-20:
Have you built the rolling stack from source? We recently changed this with https://github.com/ros-controls/ros2_controllers/pull/1247, but the binaries haven't been synced yet.

mhubii, 2024-11-20:
hi @christophfroehlich and thank you for the hint. Been using distributed binaries

mhubii, 2024-11-20:
I presume that this is fixed, just not in the binaries yet. Closing.

closed 2024-11-20 by mhubii

bmagyar, 2024-11-21:
Correct. It was flagged up by the tests straight away. We still support getting the description from params but it's only for backward compatibility for older distros.