ros-controls/ros2_controllers #1380: mecanum_driver_controller fails when built from source

Mechazo11, 2024-11-22:
Hi, 

I have clone the latest ```master``` branch of ros2_controllers for a ROS 2 Jazzy workspace that I am building from source. I am getting the following error. Please advise how to solve this problem.

```bash
--- stderr: mecanum_drive_controller                                                        
/home/tigerwife/ubuntu22_jazzy_ws/src/ros-controls/ros2_controllers/mecanum_drive_controller/src/mecanum_drive_controller.cpp: In member function ‘virtual controller_interface::CallbackReturn mecanum_drive_controller::MecanumDriveController::on_deactivate(const rclcpp_lifecycle::State&)’:
/home/tigerwife/ubuntu22_jazzy_ws/src/ros-controls/ros2_controllers/mecanum_drive_controller/src/mecanum_drive_controller.cpp:316:39: error: void value not ignored as it ought to be
  316 |       command_interfaces_[i].set_value(std::numeric_limits<double>::quiet_NaN());
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/tigerwife/ubuntu22_jazzy_ws/src/ros-controls/ros2_controllers/mecanum_drive_controller/src/mecanum_drive_controller.cpp: In member function ‘virtual controller_interface::return_type mecanum_drive_controller::MecanumDriveController::update_and_write_commands(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/tigerwife/ubuntu22_jazzy_ws/src/ros-controls/ros2_controllers/mecanum_drive_controller/src/mecanum_drive_controller.cpp:447:48: error: could not convert ‘(&((mecanum_drive_controller::MecanumDriveController*)this)->mecanum_drive_controller::MecanumDriveController::<anonymous>.controller_interface::ChainableControllerInterface::<anonymous>.controller_interface::ControllerInterfaceBase::command_interfaces_.std::vector<hardware_interface::LoanedCommandInterface>::operator[](((std::vector<hardware_interface::LoanedCommandInterface>::size_type)mecanum_drive_controller::MecanumDriveController::FRONT_LEFT)))->hardware_interface::LoanedCommandInterface::set_value(((double)wheel_front_left_vel))’ from ‘void’ to ‘bool’
  447 |       command_interfaces_[FRONT_LEFT].set_value(wheel_front_left_vel) &&
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/home/tigerwife/ubuntu22_jazzy_ws/src/ros-controls/ros2_controllers/mecanum_drive_controller/src/mecanum_drive_controller.cpp:448:49: error: could not convert ‘(&((mecanum_drive_controller::MecanumDriveController*)this)->mecanum_drive_controller::MecanumDriveController::<anonymous>.controller_interface::ChainableControllerInterface::<anonymous>.controller_interface::ControllerInterfaceBase::command_interfaces_.std::vector<hardware_interface::LoanedCommandInterface>::operator[](((std::vector<hardware_interface::LoanedCommandInterface>::size_type)mecanum_drive_controller::MecanumDriveController::FRONT_RIGHT)))->hardware_interface::LoanedCommandInterface::set_value(((double)wheel_front_right_vel))’ from ‘void’ to ‘bool’
  448 |       command_interfaces_[FRONT_RIGHT].set_value(wheel_front_right_vel) &&
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
/home/tigerwife/ubuntu22_jazzy_ws/src/ros-controls/ros2_controllers/mecanum_drive_controller/src/mecanum_drive_controller.cpp:458:75: error: could not convert ‘(&((mecanum_drive_controller::MecanumDriveController*)this)->mecanum_drive_controller::MecanumDriveController::<anonymous>.controller_interface::ChainableControllerInterface::<anonymous>.controller_interface::ControllerInterfaceBase::command_interfaces_.std::vector<hardware_interface::LoanedCommandInterface>::operator[](((std::vector<hardware_interface::LoanedCommandInterface>::size_type)mecanum_drive_controller::MecanumDriveController::FRONT_LEFT)))->hardware_interface::LoanedCommandInterface::set_value(0.0)’ from ‘void’ to ‘bool’
  458 |     const bool value_set_error = command_interfaces_[FRONT_LEFT].set_value(0.0) &&
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/tigerwife/ubuntu22_jazzy_ws/src/ros-controls/ros2_controllers/mecanum_drive_controller/src/mecanum_drive_controller.cpp:459:76: error: could not convert ‘(&((mecanum_drive_controller::MecanumDriveController*)this)->mecanum_drive_controller::MecanumDriveController::<anonymous>.controller_interface::ChainableControllerInterface::<anonymous>.controller_interface::ControllerInterfaceBase::command_interfaces_.std::vector<hardware_interface::LoanedCommandInterface>::operator[](((std::vector<hardware_interface::LoanedCommandInterface>::size_type)mecanum_drive_controller::MecanumDriveController::FRONT_RIGHT)))->hardware_interface::LoanedCommandInterface::set_value(0.0)’ from ‘void’ to ‘bool’
  459 |                                  command_interfaces_[FRONT_RIGHT].set_value(0.0) &&
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
gmake[2]: *** [CMakeFiles/mecanum_drive_controller.dir/build.make:76: CMakeFiles/mecanum_drive_controller.dir/src/mecanum_drive_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:208: CMakeFiles/mecanum_drive_controller.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
```

**To Reproduce**
Steps to reproduce the behavior: Source build the latest ```master``` branch of the  ```ros2_controller``` package

```bash
git clone -b master --single-branch https://github.com/ros-controls/ros2_controllers.git
colcon build
```

**Expected behavior**
```ros2_controllers``` and all its subpackages builds without fail.

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Version 22.04
- Building it from source as part of a ROS 2 Jazzy workspace in Ubuntu 22.04

**Additional context**
It appears the `mecanum_drive_controller` package was added just a few days ago (about 3-4 days from the time I posted this issue) and CI is failing. I think if the package is unstable it should be removed from `master` branch until it is fixed.

christophfroehlich, 2024-11-22:
Have you built ros2_control from source? Use this [repos file](https://github.com/ros-controls/ros2_controllers/blob/master/ros2_controllers.jazzy.repos) for your workspace, and build the whole stack from source. It might be necessary that you uninstall older binary packages before.

Mechazo11, 2024-11-22:
Hi @christophfroehlich I think I found the problem. Apparently, my copy of```ros2_control``` package somehow ended up with the wrong file for ```loaned_command_interface.hpp``` which previously defined ```set_value``` method as a void instead of the current implementation that should make it a boolean. Will report back if downloading the correct copy of ```ros2_control``` fixed the problem

Mechazo11, 2024-11-22:
Redownloading the latest ```master``` branch of ```ros2_control``` package solved this problem. Thank you @christophfroehlich for pointing me to the repos file.

closed 2024-11-22 by Mechazo11