ros-controls/ros2_controllers #1409: High CPU load due to joint state broadcaster

firesurfer, 2024-12-05:
**Describe the bug**

I am currently looking into the issue why our ros2control node is consuming 100% cpu load. 
For testing purposes I disabled all loaded controllers one by one. As expected the cpu drops a bit for each controller. But for the joint state broadcaster this was rather extreme. I see a change in cpu load for node of ~15-20%. 
This is not the case for other controllers that publish messages at the same rate of 500Hz. 

We have a rather large amount of joints in our system: 34. 

**Expected behavior**

I would expect that such a simple controller shouldn't consume that much cpu power. 

**Environment (please complete the following information):**
 - OS:  ubuntu 24.04
 - Version Jazzy - 4.16.0

**Additional context**

When taking a look at the code I would guess at the moment three places that could result in a higher cpu load:

1. there are rather many operations on the `map_interface_to_joint_state_` -> but unordered_map should be fast enough
2. mutex operation in the RT publisher
3. Reading the values from the state interfaces -> Not sure if there is any synchronization happening ?

christophfroehlich, 2024-12-05:
Strange. To narrow that down, I'd comment the different publishers in the update() method and see if the publishers or the `state_interface.get_value()` is causing this behavior..

saikishor, 2024-12-09:
I highly doubt that it is caused due to the handles. In a synchronous system, mutexes shouldn't do anything. So, as @christophfroehlich suggested try to check with the publishers and see.

jordan-palacios, 2025-04-16:
We are currently having issues with this broadcaster too. In our case I think simple due to the sheer amount of state interfaces our setup has: ~550. 

After a brief look, the broadcaster could use some optimization. There a lot of map lookups on the `update()` function that could be cached. I'll prepare a PR.

saikishor, 2025-04-16:
> We are currently having issues with this broadcaster too. In our case I think simple due to the sheer amount of state interfaces our setup has: ~550.
>
> After a brief look, the broadcaster could use [...]

Thank you @jordan-palacios

jordan-palacios, 2025-04-16:
Performance of the current version in `master`.

```
[INFO] [1744704812.699545539] [joint_state_broadcaster]: Average update time: 240.174700 us
[INFO] [1744704812.699598892] [joint_state_broadcaster]: Variance: 231730.945780 us
--
[INFO] [1744704815.285795981] [joint_state_broadcaster]: Average update time: 236.002800 us
[INFO] [1744704815.285851910] [joint_state_broadcaster]: Variance: 297345.111392 us
--
[INFO] [1744704817.849313397] [joint_state_broadcaster]: Average update time: 234.468100 us
[INFO] [1744704817.849375057] [joint_state_broadcaster]: Variance: 231589.582582 us
--
[INFO] [1744704820.471171285] [joint_state_broadcaster]: Average update time: 239.791200 us
[INFO] [1744704820.471218683] [joint_state_broadcaster]: Variance: 462334.572003 us
--
[INFO] [1744704823.045886822] [joint_state_broadcaster]: Average update time: 235.594400 us
[INFO] [1744704823.045938966] [joint_state_broadcaster]: Variance: 26580.964489 us
--
[INFO] [1744704825.625277357] [joint_state_broadcaster]: Average update time: 235.583400 us
[INFO] [1744704825.625330096] [joint_state_broadcaster]: Variance: 231611.694044 us
--
[INFO] [1744704828.254401907] [joint_state_broadcaster]: Average update time: 241.029100 us
[INFO] [1744704828.254448233] [joint_state_broadcaster]: Variance: 462391.610853 us
--
[INFO] [1744704830.802185520] [joint_state_broadcaster]: Average update time: 232.373100 us
[INFO] [1744704830.802232534] [joint_state_broadcaster]: Variance: 811.116296 us
--
[INFO] [1744704833.448850699] [joint_state_broadcaster]: Average update time: 242.778300 us
[INFO] [1744704833.448903136] [joint_state_broadcaster]: Variance: 462345.285149 us
--
[INFO] [1744704836.060410410] [joint_state_broadcaster]: Average update time: 238.798000 us
[INFO] [1744704836.060460679] [joint_state_broadcaster]: Variance: 231659.275996 us
```

PR #1640 (2025-04-18): Fix joint_state_broadcaster performance issues
This PR will include adding a new test for checking performance in a setup with lots of state interfaces (>500). I'll include here the results of each incremental performance fix commit added.

Since we have also noticed a lot of variance in the results, we'll run it a few times. All results will be summarized here.

Fixes #1409.
closed 2025-04-18 by saikishor