ros-controls/ros2_controllers #1425: Add a controller for  afour wheel independent steering / drive mobile base, aka a swerve drive

pvandervelde, 2024-12-15:
I would like to add a controller for a 4 wheel independent steering and independent drive (aka swerve drive) mobile base. In this kind of mobile base each wheel has two degrees of freedom, one for moving the wheel around its axle and one for the steering motion of the wheel. Each wheel is independently controllable from the other wheels. 

There have been several attempts at creating ROS controllers for such a system, e.g. https://github.com/ros-controls/ros_controllers/pull/441, https://github.com/MarkNaeem/ros_controllers/tree/noetic-devel/swerve_steering_controller and https://github.com/ros-controls/ros2_controllers/pull/1148.

I have created a [proof of concept](https://github.com/pvandervelde/zinger_swerve_controller) for one such drive in python. Additionally I have created a [simple simulation tool](https://github.com/pvandervelde/basic-swerve-sim) to test different control algorithms.

If there is interest I would like to implement a swerve controller for ROS2 control.

christophfroehlich, 2024-12-15:
Hi @pvandervelde!
Your contribution would be highly appreciated. I see that you started documenting the kinematics, as a first step you could add your documentation [here](https://github.com/ros-controls/ros2_controllers/blob/master/doc/mobile_robot_kinematics.rst) and then build your controller upon that nomenclature.

pvandervelde, 2024-12-17:
@christophfroehlich That does sound like a good place to start. For the implementation process I was aiming to follow the process suggested in [this comment](https://github.com/ros-controls/ros2_controllers/issues/264#issuecomment-996646394) by bmagyar. I'll reference this issue in all PRs.

I might have some questions about some parts of the library. Would those best be asked in the ROS discourse or just in this issue?

Forsyth-Creations, 2025-02-19:
Hello! I would also like to work on this. What's the status on this @pvandervelde? I have a fork that I just started on, but wanted to check in here first. Thought I might start to port your work in

Forsyth-Creations, 2025-02-19:
Oh yeah, I also have this. Been working on this for my masters research thesis:
https://github.com/Forsyth-Creations/ros_rigor
I have a basic controller written in python, but it for speed it would be nice to have it in C. Something I'm considering as apart of the research

christophfroehlich, 2025-02-20:
@Forsyth-Creations you are very welcome to submit a PR if @pvandervelde does not respond any more within the next days.

pvandervelde, 2025-02-20:
> Hello! I would also like to work on this. What's the status on this [@pvandervelde](https://github.com/pvandervelde)? I have a fork that I just started on, but wanted to check in here first. Thought I might start to port your work in

To be honest I have been hit with life and work and have been snowed under. I should have some spare time coming up in the next few weeks but if you have something that works then you should put it in.

Amronos, 2025-02-20:
>     1. How do you all go about testing these controllers? Is there documentation on this?

I made a Gazebo simulation to test mine and then tested it in real life with a custom robot. Also, have a look at [Mock Components](https://control.ros.org/rolling/doc/ros2_control/hardware_interface/doc/mock_components_userdoc.html)

>     2. How is this controller subsequently used for something like Gazebo? Is there a plugin schema I would need to write?

Have a look at the [gz_ros2_control documentation](https://control.ros.org/rolling/doc/gz_ros2_control/doc/index.html).

christophfroehlich, 2025-02-20:
1. We write also unit tests to ensure everything works properly.
2. I suggest that you have a look at the already available controllers, like the mecanum_drive_controller, and use this as a template.

Forsyth-Creations, 2025-02-26:
Sweet, thank you! 

Built this last night as a test of the kinematics:

https://www.forsythcreations.com/swerve_drive

christophfroehlich, 2026-07-04:
you mean the admissible velocity configuration?

This would be useful in all controllers, we currently only support limiting the twist but don't consider any joint limits (is the twist admissible at all?). But what to do if the constraints are hit? One has to come up with a plan how to project that into the admissible region again, or should commands be rejected then?