ros-controls/ros2_controllers #1442: Cleanup all the deprecations warning from the recent ros2_control variant changes

saikishor, 2024-12-22:
## Background

Recently after the new variants feature in Handles (https://github.com/ros-controls/ros2_control/pull/1688) We have a bunch of deprecation warnings that needs to be cleanup. This could be a nice first issue.

## Instructions

The best approach to this is to create the temporary variables where ever needed and then if the `set_value` or `get_value` returns `False`. Then, better to print a warning and simply return OK. This way, if we have some failures in getting data, the user will be able to know and can introspect properly. 

Moreover, If everything is running synchronously. it should work as expected without any failures.

For example:

For the following lines of the code in the diff_drive_controller:
https://github.com/ros-controls/ros2_controllers/blob/81c0d41e4feebfa7e7c81b01d0ac1101eaf10e5d/diff_drive_controller/src/diff_drive_controller.cpp#L268-L272

The expected code should look like:

```
  for (size_t index = 0; index < static_cast<size_t>(wheels_per_side_); ++index)
  {
    if (!registered_left_wheel_handles_[index].velocity.get().set_value(velocity_left) ||
    !registered_right_wheel_handles_[index].velocity.get().set_value(velocity_right))
    {
      RCLCPP_WARN(get_node()->get_logger(), "Error while setting the velocity_left : %f "
        "and velocity_right : %f to the command handles", velocity_left, velocity_right);
      return controller_interface::return_type::OK;
    }
  }
```

For the following `get_value` code in the diff_drive_controller:
https://github.com/ros-controls/ros2_controllers/blob/81c0d41e4feebfa7e7c81b01d0ac1101eaf10e5d/diff_drive_controller/src/diff_drive_controller.cpp#L162-L164

The expected outcome would be:

```
bool left_success = false;
const double left_feedback = [&]() {
    double temp;
    left_success = registered_left_wheel_handles_[index].feedback.get().get_value(temp);
    return temp;
}();

bool right_success = false;
const double right_feedback = [&]() {
    double temp;
    right_success = registered_right_wheel_handles_[index].feedback.get().get_value(temp);
    return temp;
}();

// Now you can use left_success and right_success to check if get_value was successful
if (!left_success || !right_success) {
    RCLCPP_WARN(get_node()->get_logger(), "Error while fetching information from the state interfaces");
    return controller_interface::return_type::OK;
}
```

or
```

double left_feedback, right_feedback; 
if(!registered_left_wheel_handles_[index].feedback.get().get_value(left_feedback) ||
    !registered_right_wheel_handles_[index].feedback.get().get_value(right_feedback))
{
  RCLCPP_WARN(get_node()->get_logger(), "Error while fetching information from the state interfaces");
  return controller_interface::return_type::OK;
}
```

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_controllers`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_controllers` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

saikishor, 2024-12-22:
@kumar-sanjeeev or @MartinPeris you might be interested to work on this

kumar-sanjeeev, 2024-12-22:
@saikishor, thanks. I will look into this.

kumar-sanjeeev, 2024-12-22:
@saikishor Hi,

I took a look at this and made adjustments to the `diff_drive_controller` code to capture and handle the return values of the `get_value` and `set_value` where ever used. I think similar changes need to be applied to the other controllers as well? If so, do you think I should submit one PR covering all controllers, or would it be better to start with a PR for diff_drive_controller and then add commits for the others later? Let me know what you think.

saikishor, 2024-12-22:
@kumar-sanjeeev let's try to address all these depreciations in a single PR

christophfroehlich, 2024-12-22:
but you always can submit a draft PR if you want to get feedback before you work on the other controllers

kumar-sanjeeev, 2024-12-22:
@christophfroehlich, yeah I would like to do that, just to check if I am in right direction or not. So I will submit draft PR.

saikishor, 2024-12-22:
> @christophfroehlich, yeah I would like to do that, just to check if I am in right direction or not. So I will submit draft PR.

Thank you!

kumar-sanjeeev, 2025-03-02:
@saikishor, hi, it would be nice if you can give me starting point for one controller (suggestions/changes on one controller). After that I can takeover for rest of them.

saikishor, 2025-03-02:
> @saikishor, hi, it would be nice if you can give me starting point for one controller (suggestions/changes on one controller). After that I can takeover for rest of them. 
> 

Sure, will do

saikishor, 2025-03-02:
@kumar-sanjeeev I made the changes for diff_drive_controller in #1565. Please take a look and let you know, if you can make the changes for the rest of the controllers :)

kumar-sanjeeev, 2025-03-02:
@saikishor, thanks for building blueprint for changes. I will take a look and get back to you.

kumar-sanjeeev, 2025-03-02:
@saikishor, I looked at #1565 . To me, it seems doable for me to implement similar in other controllers wherever  needed. Would you want me to apply changes for each controller in a separate PR, or should I create a single PR for the rest of the controllers?

saikishor, 2025-03-03:
Make one PR for all the changes. 
I suggest you to make the changes for 1 or 2 packages and let us review it, once it looks good, then apply them :)

closed 2026-01-09 by saikishor