ros-controls/ros2_controllers #1453: use_stamped_vel parameter is missed in Diff_drive_controller in Jazzy

drfenixion, 2024-12-29:
In current master (Jazzy) branch use_stamped_vel parameter is missed
[https://github.com/ros-controls/ros2_controllers/blob/master/diff_drive_controller/src/diff_drive_controller_parameter.yaml](https://github.com/ros-controls/ros2_controllers/blob/master/diff_drive_controller/src/diff_drive_controller_parameter.yaml)

It present in iron branch
[https://github.com/ros-controls/ros2_controllers/blob/iron/diff_drive_controller/src/diff_drive_controller_parameter.yaml#L109](https://github.com/ros-controls/ros2_controllers/blob/iron/diff_drive_controller/src/diff_drive_controller_parameter.yaml#L109)

Therefore it is imposible to use use_stamped_vel in Jazzy. I tested it in Iron and Jazzy.

Is it bug or feature cut?

saikishor, 2024-12-29:
Hello @drfenixion,

From Jazzy on, the default message type is **TwistStamped**.

https://github.com/ros-controls/ros2_controllers/blob/f23b7d833a41bff6e2125ff5243e0ed95133e6d8/diff_drive_controller/src/diff_drive_controller.cpp#L405-L424

What is the exact issue that you are facing?

drfenixion, 2024-12-29:
@saikishor , issue is - i cant switch to cmd_vel_unstamped topic because there is not present "use_stamped_vel" parameter in ros2_controllers master branch. I can use only cmd_vel (stamped).

In Iron branch i can switch between "unstamped" and "stamped" cmd_vel topics by "use_stamped_vel" parameter.

saikishor, 2024-12-29:
Hello @drfenixion!

This is a breaking change from Iron to Jazzy. It is already specified in the migration notes : https://github.com/ros-controls/ros2_controllers/blob/master/doc/migration.rst#diff_drive_controller. From Jazzy on, the diff_drive_controller only supports TwistStamped type. You will need to migrate parts of your node to work with this new message type.

Thank you.

closed 2024-12-29 by saikishor

christophfroehlich, 2024-12-29:
For reference, nav2 also uses TwistStamped
https://discourse.ros.org/t/notice-nav2-migrated-to-twiststamped-to-replace-twist-for-cmd-vel-topics/40944