ros-controls/ros2_controllers #1455: Discontinuity in FollowJointTrajectory Velocity in ROS 2

abuibaid, 2024-12-29:
Hi Everyone,

I’m currently working on controlling a UR5e robot using ROS. Initially, I used ROS Noetic (ROS 1), which was quite straightforward. However, with the End-of-Life (EOL) for ROS 1 approaching, we’ve started migrating to ROS 2, which is fine overall.

Here’s the issue I’m encountering in ROS 2 (simulation): I’m sending the robot a series of random trajectories (position and velocity) using an ActionClient. Each trajectory is designed to take 6 seconds to reach its goal, but my goal is to switch trajectories before the 6-second mark. In ROS 1 (Noetic), this was handled smoothly—there was a seamless transition between trajectories, even if the previous trajectory hadn’t reached its goal.

In ROS 2, however, while the position states show good results, the velocity states exhibit noticeable discontinuities. This kind of behavior was not present in ROS 1, where transitions between trajectory commands were much smoother. The following figure illustrates the discontinuity issue I’m facing.

[image]
[image]

The code that i have used is also provided.

        import numpy as np
        from control_msgs.action import FollowJointTrajectory
        from trajectory_msgs.msg import JointTrajectoryPoint
        import time
        import rclpy
        from rclpy.node import Node
        from rclpy.duration import Duration
        from rclpy.action import ActionClient
      
      class robot_action(Node):
          def __init__(self):
              super().__init__('ur5e_robotiq_controller')
              self.intialization_robot_control()
      
      
          def intialization_robot_control(self):
              # Create action client
              self.ur5e_action_client=ActionClient(self,action_type=FollowJointTrajectory,action_name='/joint_trajectory_controller/follow_joint_trajectory')
      
              self.create_timer(0.3, self.control_loop_callback)  # Update every 0.1 seconds
      
              
          def transitionRobot(self,duration,trajectory):
              # Create points for the trajectory
              self.point = JointTrajectoryPoint()
              self.point.time_from_start = Duration(seconds = (duration[1]-duration[0])).to_msg()
              self.point.positions = trajectory[0:6].tolist()#[0.0, -1.57, 1.57, 0.0, 1.57, 0.0]  # Example joint positions (radians)
              self.point.velocities = trajectory[6:12].tolist()#[0.0, 0, 0, 0.0, 0, 0.0]  # Example joint positions (radians)
              self.point.accelerations =  [0.01, 0.01, 0.01, 0.01, 0.01, 0.01]  # Example accelerations
              # Create ur5e message for transition Robot:
              self.msg_ur5e = FollowJointTrajectory.Goal()
              self.msg_ur5e.trajectory.joint_names = ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
              # self.msg_ur5e.trajectory.header.stamp = Time(sec=0, nanosec=0) #self.get_clock().now().to_msg()
      
              self.msg_ur5e.trajectory.points=[self.point]
              # Action client
      
      
              self.get_logger().info(f"Trajectory joint names: {self.msg_ur5e.trajectory.joint_names}")
              for i, point in enumerate(self.msg_ur5e.trajectory.points):
                  self.get_logger().info(
                      f"Point {i}: positions={point.positions}, "
                      f"velocities={point.velocities if point.velocities else 'None'}, "
                      f"accelerations={point.accelerations if point.accelerations else 'None'}, "
                      f"time_from_start={point.time_from_start.sec}.{point.time_from_start.nanosec}"
                  )
              self.ur5e_action_client.wait_for_server()
              # self.msg_ur5e.trajectory.header.stamp = self.get_clock().now().to_msg()
              goal_future =self.ur5e_action_client.send_goal_async(self.msg_ur5e)
              goal_future.add_done_callback(self.goal_response_callback)
          
          def goal_response_callback(self, future):
              goal_handle = future.result()
              if not goal_handle.accepted:
                  self.get_logger().info('Goal rejected :(')
                  return
              
              self.get_logger().info('Goal accepted :)')
              self._get_result_future = goal_handle.get_result_async()
              self._get_result_future.add_done_callback(self.get_result_callback)
      
          def get_result_callback(self, future):
              result = future.result().result
              self.get_logger().info('Result: {0}'.format(result))
              # rclpy.shutdown()
      
      
          def feedback_callback(self, feedback_msg):
              self.get_logger().info('Received feedback: {0}'.format(feedback_msg.feedback))
              return
      
      
          
          def control_loop_callback(self) -> None:
                  q1 = [-3,-1,1,3,1,-1,-3,-1,1,3,1,-1]
                  position = np.zeros(6)
                  position[0] = q1[0]
                  velocity = np.ones(6)*0.01
                  self.transitionRobot([0,6],np.concatenate([position,velocity],axis=0))
                  time.sleep(1.5)  # Wait for arm to reach target (2.5s)
          
                  q1 = [-3,-1,1,3,1,-1,-3,-1,1,3,1,-1]
                  position[0] = q1[1]
                  self.transitionRobot([0,6],np.concatenate([position,velocity],axis=0))
                  time.sleep(1.5)  # Pause for 1 second at target
          
                  q1 = [-3,-1,1,3,1,-1,-3,-1,1,3,1,-1]
                  position[0] = q1[2]
                  self.transitionRobot([0,6],np.concatenate([position,velocity],axis=0))
                  time.sleep(1.5)  # Pause for 1 second at target
          
                  q1 = [-3,-1,1,3,1,-1,-3,-1,1,3,1,-1]
                  position[0] = q1[3]
                  self.transitionRobot([0,6],np.concatenate([position,velocity],axis=0))
                  time.sleep(1.5)  # Pause for 1 second at target
          
                  q1 = [-3,-1,1,3,1,-1,-3,-1,1,3,1,-1]
                  position[0] = q1[4]
                  self.transitionRobot([0,6],np.concatenate([position,velocity],axis=0))
                  time.sleep(1.5)  # Pause for 1 second at target
          
                  q1 = [-3,-1,1,3,1,-1,-3,-1,1,3,1,-1]
                  position[0] = q1[5]
                  self.transitionRobot([0,6],np.concatenate([position,velocity],axis=0))
                  time.sleep(1.5)  # Pause for 1 second at target
          
                  # Final pause before next cycle
                  time.sleep(0.3)
                  self.get_logger().info("Control loop callback executed.")
      
      def main(args=None):
          rclpy.init(args=args)
          robot = robot_action()
      
          try:
              rclpy.spin(robot)
          except KeyboardInterrupt:
              pass
          finally:
              robot.destroy_node()
              rclpy.shutdown()
      
          # Shutdown
          robot.destroy_node()
          rclpy.shutdown()
      
      if __name__ == '__main__':
          main()

have any of you witness such a behaviour? how to solve this issue?

thanks for the help in advance.

Regards,

christophfroehlich, 2024-12-30:
I think what you are asking for is a proper replacement of old trajectories? this hasn't been ported to ROS 2 yet https://github.com/ros-controls/ros2_controllers/issues/84

abuibaid, 2024-12-30:
Hi @christophfroehlich, yes are right :(. Anyway thank you for replying me and looking forward to see this working

saikishor, 2024-12-30:
Hello @abuibaid,

We currently don't have a timeline for this. If you are interested in taking over and please open a pull request. We can get it reviewed and merged.

Thank you

closed 2024-12-30 by christophfroehlich

bmagyar, 2024-12-30:
> ID: <ros-controls/ros2_controllers/issue/1455/issue_event/15783572354@
> github.com>

We should put this on the roadmap for next year

abuibaid, 2025-01-07:
Hi everyone, speaking of this issue, which parameter of the joint_trajectory_controller should take care of this issue? is it "allow_integration_in_goal_trajectories"? I would like to see how to make it work because it is urgent for me :D

christophfroehlich, 2025-01-07:
There is no parameter yet. This feature has to be implemented or ported from ros_control, see the link in the PR above.

pastoriomarco, 2025-05-29:
If it may be of some use to anyone needing trajectory replacement: using moveitcpp there may be a workaround.

- Use the current trajectory as baseline for the new trajectory
- Move the points of the trajectory in the past subtracting the current time elapsed to the time_from_start of each point
- Delete all points in the future
- Add the new trajectory and send it to the JTC