ros-controls/ros2_controllers #1456: Extend tests for odometry of steering controllers library

christophfroehlich, 2024-12-29:
We had an issue (fixed with #1454) where the odometry was wrong in case of a reference timeout.

There should be a test for this behavior, maybe in the test [test_both_update_methods_for_ref_timeout](https://github.com/ros-controls/ros2_controllers/blob/f23b7d833a41bff6e2125ff5243e0ed95133e6d8/steering_controllers_library/test/test_steering_controllers_library.cpp#L82)

AdityaPawar162, 2025-03-24:
Hi @christophfroehlich ,
I would like to work on this issue.

AdityaPawar162, 2025-03-24:
Hi @christophfroehlich,
From what i understood, I need to extend the existing the test for odometry behaviour during reference timeout. To do that i need to,
1. Track odometry values before and after the timeout.
2. Verify that position values remain stable and don't get incorrectly updated.
3. Check that velocity values in odometry reset to zero.

Am i heading in the right direction?