ros-controls/ros2_controllers #1461: Refactor odometry class of steering controllers library

christophfroehlich, 2025-01-01:
## Background

The `SteeringOdometry` class handles forward and inverse kinematics. 
This leads to code like this one, which is confusing (why should `odometry` be used for calculating commands).
```
    auto [traction_commands, steering_commands] = odometry_.get_commands(
      reference_interfaces_[0], reference_interfaces_[1], params_.open_loop,
      params_.reduce_wheel_speed_until_steering_reached);
```

Suggestion: Split the class in forward and inverse kinematics, maybe inheriting from a common parameter class.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_controllers/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_controllers`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_controllers` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

christophfroehlich, 2025-01-08:
What I haven't mentioned in the first idea there: We have to deprecate the old library, because it is released into a stable distro (Jazzy) and we can't break API without notice there.

1. The easiest solution is just to rename the class SteeringOdometry (incl. namespace and filenames) to something including FK and IK, maybe just steering_kinematics. Then add an alias in the old filenames including a compile-time warning that this is deprecated, see [examples for this](https://github.com/ros-controls/realtime_tools/blob/master/include/realtime_tools/realtime_box.h).
2. Same can be done with splitting SteeringOdometry into two, but maybe this is not necessary if the naming is clear.

bhagyeshagresar, 2025-03-23:
Hello @christophfroehlich and @nakul-py ! Is this issue still open for contribution?

christophfroehlich, 2025-03-23:
The linked PR #1475 is stale, @bhagyeshagresar you are welcome to make your proposal. Please consider my comment above, and the comments in the linked PR. Could you maybe describe your proposal first in this issue, so we can discuss it before you start implementing it?

bhagyeshagresar, 2025-03-25:
Hello @christophfroehlich. I was gonna follow along with your comments for the solution but let me take some time to go through it. I need to get a little more familiar with the project.

christophfroehlich, 2025-03-25:
The simplest approach would be just to rename `odometry` to `kinematics`, which is then valid for forward kinematics (odometry calculation) and inverse kinematics (getting commands).
For deprecation, I suggest renaming the files, adding a new steering_odometry.hpp file pointing to the new header and with a compile time warning if it is used. In this file an alias for the symbols like SteeringOdometry to SteeringKinematics will do the job.