ros-controls/ros2_controllers #1500: Error while building the package in Dockerfile

Abhishek8857, 2025-01-19:
**Describe the bug**

 1831.9 Starting >>> diff_drive_controller
1843.6 --- stderr: diff_drive_controller
1843.6 In file included from /colcon_ws/src/ros2_controllers/diff_drive_controller/src/diff_drive_controller.cpp:25:
1843.6 /colcon_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/diff_drive_controller.hpp:45:10: fatal error: diff_drive_controller/diff_drive_controller_parameters.hpp: No such file or directory
1843.6    45 | #include "diff_drive_controller/diff_drive_controller_parameters.hpp"
1843.6       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1843.6 compilation terminated.
1843.6 gmake[2]: *** [CMakeFiles/diff_drive_controller.dir/build.make:76: CMakeFiles/diff_drive_controller.dir/src/diff_drive_controller.cpp.o] Error 1
1843.6 gmake[2]: *** Waiting for unfinished jobs....
1843.6 gmake[1]: *** [CMakeFiles/Makefile2:206: CMakeFiles/diff_drive_controller.dir/all] Error 2
1843.6 gmake[1]: *** Waiting for unfinished jobs....
1843.6 gmake: *** [Makefile:146: all] Error 2
1843.6 ---
1843.6 Failed   <<< diff_drive_controller [11.7s, exited with code 2]
1843.6 
1843.6 Summary: 49 packages finished [30min 42s]
1843.6   1 package failed: diff_drive_controller
1843.6   13 packages had stderr output: controller_manager diff_drive_controller kinova_vision moveit_configs_utils picknik_reset_fault_controller picknik_twist_controller realtime_tools ros2_controllers_test_nodes ros_gz_bridge ros_ign_bridge ros_ign_gazebo ros_ign_image rqt_joint_trajectory_controller
1843.6   70 packages not processed
------
Dockerfile:301
--------------------
 300 |     # Build the workspace with resource management
 301 | >>> RUN source /opt/ros/humble/setup.bash && \
 302 | >>>     MAKEFLAGS="-j4 -l2" colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 3 --symlink-install --executor sequential
 303 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c source /opt/ros/humble/setup.bash &&     MAKEFLAGS=\"-j4 -l2\" colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 3 --symlink-install --executor sequential" did not complete successfully: exit code: 2

**Expected behavior**
Supposed to build without errors. Dockerfile has ros2_kortex, ros2_kortex_vision, moveit2, moveit_task_constructor as submodules

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Linux
 - Version 22.04

**Additional context**
using these repositories as well

`repositories:
  picknik_controllers:
    type: git
    url: https://github.com/picknikrobotics/picknik_controllers.git
    version: main

  ros2_robotiq_gripper:
    type: git
    url: https://github.com/picknikrobotics/ros2_robotiq_gripper.git
    version: b6136bdc866a929bfb096890ca61dde1335ffd30

  serial:
    type: git
    url: https://github.com/tylerjw/serial.git
    version: ros2

  realtime_tools:
    type: git
    url: https://github.com/ros-controls/realtime_tools.git
    version: humble

  ros2_controllers:
    type: git
    url: https://github.com/ros-controls/ros2_controllers.git
    version: humble

  control_msgs:
    type: git
    url: https://github.com/ros-controls/control_msgs.git
    version: humble

  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control.git
    version: humble

  ros2_robotiq_gripper:
    type: git
    url: https://github.com/picknikrobotics/ros2_robotiq_gripper.git
    version: b6136bdc866a929bfb096890ca61dde1335ffd30

  gazebo_ros2_control:
    type: git
    url: https://github.com/ros-simulation/gazebo_ros2_control.git
    version: humble

  gz_ros2_control:
    type: git
    url: https://github.com/ros-controls/gz_ros2_control.git
    version: humble

  ros_gz:
    type: git
    url: https://github.com/gazebosim/ros_gz.git
    version: humble

`

bmagyar, 2025-01-19:
I think this may be related to the recent move of the generate parameter
library path. You could try pulling that library from source too or going
back a few commits back on ros2_control.

christophfroehlich, 2025-01-19:
Exactly, will solve itself with the upcoming humble sync, or you use `testing` binaries or you compile generate_parameter_library from source.

Juliaj, 2025-01-20:
@Abhishek8857 , in case you want to try `testing` binaries option,  here is what I did to resolve the compilation errors from generate parameter library on my local dev env. 

- add `ros2-testing` to `ros2.list` .
```
# deb http://packages.ros.org/ros2/ubuntu noble main
deb http://packages.ros.org/ros2-testing/ubuntu noble main
```
- update the apt index:
```
sudo apt update
```

- pull  `generate-parameter-library` from the testing repository, for example:

```
sudo apt install ros-jazzy-generate-parameter-library
```
- switch back to `ros2` for packages

You can find more details at https://docs.ros.org/en/jazzy/Installation/Testing.html.

christophfroehlich, 2025-01-21:
This should be fixed now with https://discourse.ros.org/t/new-packages-for-humble-hawksbill-2025-01-21/41658

closed 2025-01-21 by christophfroehlich