ros-controls/ros2_controllers #1505: [JTC] Parameter open_loop_control should be the default

Thieso, 2025-01-24:
**Is your feature request related to a problem? Please describe.**
I am doing trajectory tracking with a Kinova Arm and I send a new trajectory every 0.1s. This only works with `open_loop_control` set to `true` because otherwise the current state is taken as the start for the interpolation. This is weird to me as it is always desired to update from the current desired state and not from the actual state as you are updating the desired trajectory and not the actual trajectory. The controller is also still doing closed loop control with respect to the desired state right? 

Can you explain the thought behind updating from the actual state and why this is useful? Maybe I am mistaken. 

**Describe the solution you'd like**

I think the `open_loop_control` parameter should be `true` by default and maybe renamed as there still is a closed control loop if you set it to true. 

In case I am mistaken, feel free to clear things up :) Thanks for your work.

christophfroehlich, 2025-02-01:
Hi Thies!

It uses the last desired state only if a new trajectory is sampled for the first time, i.e., sent via an ROS action
https://github.com/ros-controls/ros2_controllers/blob/694d6f184b91d7d4c6f52052e298aa106a219253/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L195-L204
Later on, the current state is not involved in interpolation.

Howerver, closed-loop PID control is also deactivated if open_loop=true
https://github.com/ros-controls/ros2_controllers/blob/694d6f184b91d7d4c6f52052e298aa106a219253/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L719-L724

Maybe it makes sense to rename `open_loop` to `interpolate_from_desired_state` and just remove the open_loop from the `use_closed_loop_pid_adapter_` (if one wants to deactivate it, just set the gains to zero).

Thieso, 2025-02-04:
Ah I see, I was not using actions but just the publisher, in that case it will set this every time I think. But this is then my bad. 
Otherwise, I agree with you and think the solution you propose is good. I will make a PR

Thanks for your help.

PR #1525 (2025-03-07): [JTC] Rename open_loop_control parameter and slightly change its scope
Resolves #1505
closed 2025-03-07 by christophfroehlich