ros-controls/ros2_controllers #1516: How to perform a gradual stop with the Joint Trajectory Controller

jacob-kruse, 2025-02-03:
Hello,

I am utilizing the Joint Trajectory Controller along with the MoveIt2 Python API to motion plan with a Kuka LBR iisy 3 r760 robotic manipulator. I was wondering if there was a way to interrupt the robot while it is executing a trajectory and have it come to a gradual stop. Basically, I am trying to execute a controlled stop on command without having to use the Emergency Stop (effective, but rough on the robot). I have tried cancelling the goal and have found that although this works, it comes to an immediate, abrupt stop which I am trying to avoid. Any suggestions?

Jacob Kruse

jacob-kruse, 2025-02-03:
> This is currently not possible out-of-the box.
> 
> But soonish we will have joint-limit enforcement directly from the resource_manager of ros2_control, see [ros-controls/ros2_control#1989](https://github.com/ros-controls/ros2_control/pull/1989) This should at least limit the output to the jerk and acceleration limits of the URDF (from the ros2_control tag because there are no such tags in the original URDF spec, see [ros-controls/ros2_control#1472](https://github.com/ros-controls/ros2_control/pull/1472)).
> 
> Could you compile the version of ros2_control from the PR from source, adapt your ros2_control settings in the URDF, and test if this improves your behavior? There is no documentation of that feature yet, have a look at the URDF description of the tests and you will see the desired syntax.

Thanks for the response! I will try and implement this and let you know the results.

jacob-kruse, 2025-02-04:
@christophfroehlich 
Just to make sure I understand, by compiling the PR you tagged, I have the ability to set acceleration and jerk limits. Does that mean when I cancel a goal or stop the execution of the trajectory, the controller should abide by these acceleration and jerk limits and come to a gradual stop? If so, is the proper course of action to cancel the goal through the action server?

christophfroehlich, 2025-02-04:
cancelling from the action client, that is one possibility yes. ros2_control does not have error/emergency stop handling yet.

christophfroehlich, 2025-02-04:
I see your point. @saikishor can the joint_limits plugin consider acceleration if no acceleration command_interface exists, e.g., velocity only? How can one give the parameters then?

jacob-kruse, 2025-02-07:
@christophfroehlich
I have looked into other solutions for my desired functionality. One forum talked about the `stop_trajectory_duration` parameter from the ROS version of the `joint_trajectory_controller`. Do you know if this parameter will ever come to the ROS2 version or if there is something similar already implemented?

Also, even though the robot does not have an acceleration command interface, it does have stiffness and damping command interfaces. Does the code from the PR enforce these command limits as well. If so, is there any way I could use these with the limiting method you initially suggested to simulate a gradual stop?

jacob-kruse, 2025-02-07:
I have found a workaround solution for the time being, although the stop produced is still too abrupt for my liking.

See this [discussion thread](https://github.com/orgs/moveit/discussions/3300#discussioncomment-12096479).

christophfroehlich, 2025-02-10:
> [@christophfroehlich](https://github.com/christophfroehlich) I have looked into other solutions for my desired functionality. One forum talked about the `stop_trajectory_duration` parameter from the [...]

I was not aware that this existed with ROS. You are very welcome to port this to ROS 2 and open a PR :)