ros-controls/ros2_controllers #1519: [JTC] Action goal replacement doesn't cancel the old goal in time

fmauch, 2025-02-03:
**Describe the bug**

When sending a new action goal to the JTC while there is a goal running, the replacement seems to be working fine, but the old action does not return until the new action finished.

As originally posted in https://github.com/ros-controls/ros2_controllers/pull/1517#issuecomment-2632044373

From my understanding, we call preempt_active_goal() when [a new goal is accepted](https://github.com/ros-controls/ros2_controllers/blob/694d6f184b91d7d4c6f52052e298aa106a219253/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L1175). Inside there, we use setCanceled() to set request the target state canceled. This would be handled by the next timer callback for runNonRealtime() but this is never happening, as in the same action callback we [reset the controller's timer](https://github.com/ros-controls/ros2_controllers/blob/694d6f184b91d7d4c6f52052e298aa106a219253/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L1195).

If I add a manual call to runNonRealtime() as part of preempt_active_goal(), I get the expected output.

Trajectory replacement seems to be working fine and as expected, however the overwritten trajectory doesn't return until the overwriting trajectory finished. Then, it reports CANCELED. I noticed, that the succeed event is the first from the runNonRealtime() function that gets triggered, which is why this probably triggers the old action being canceled in the underlying (non-realtime) action-server.

This might be potentially related to #1241 

**To Reproduce**
Steps to reproduce the behavior:
1. Launch a robot with the JTC running (can also be mock hw), e.g. 
   `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e use_mock_hardware:=true initial_joint_controller:=joint_trajectory_controller robot_ip:=1234`
2. Run an action on the JTC, e.g. by `ros2 run ur_robot_driver example_move.py --ros-args -p controller_name:=joint_trajectory_controller`
3. Once the action starts executing (the robot moves in RViz), run a second client in another shell with the same command
4. The robot executes the second trajectory. Once the trajectory of the second client successfully finished, the first client also returns telling that the action was canceled.

With the example move script it looks like this:

first client:
```
[INFO 2025-02-03 23:19:43.467] [jtc_client]: Waiting for action server on joint_trajectory_controller/follow_joint_trajectory
[INFO 2025-02-03 23:19:45.470] [jtc_client]: Executing trajectory traj0
[INFO 2025-02-03 23:19:55.979] [jtc_client]: Done with result: CANCELED
[ERROR 2025-02-03 23:19:55.980] [jtc_client]: Executing trajectory failed.
```

second client:
```
INFO 2025-02-03 23:19:47.978] [jtc_client]: Waiting for action server on joint_trajectory_controller/follow_joint_trajectory
[INFO 2025-02-03 23:19:47.978] [jtc_client]: Executing trajectory traj0
[INFO 2025-02-03 23:19:55.981] [jtc_client]: Done with result: SUCCEEDED
```

**Expected behavior**
The first client should report its trajectory as preempted / aborted (to be discussed) as soon as the second client starts executing its action.

**Environment (please complete the following information):**
- Tested on Rolling master at the time of writing
- For the reproduction test having the package `ros-rolling-ur` installed should be sufficient

Mannava-Daasaradhi, 2026-09-30:
I think this was fixed by #2422 (merged 2026-06-17), which didn't reference this issue. `preempt_active_goal()` now calls `active_goal->runNonRealtime()` right after setting the result, before the `goal_handle_timer_` is reset, which is exactly the fix proposed above. The preempted goal now returns ABORTED immediately instead of CANCELED after the new trajectory finishes.

Can this be closed, or is there a remaining case (e.g. #1241 with the RT kernel) that should keep it open?

christophfroehlich, 2026-10-09:
@fmauch what do you think?

fmauch, 2026-10-10:
Yes, the issue reported here is resolved by that. What I don't particularly like (and missed when reviewing #2422) is that the result is `INVALID GOAL`. But that's for another issue. I'm closing this one.

closed 2026-10-10 by fmauch