ros-controls/ros2_controllers #1522: From Jan 15th update: `State interface with key 'wrist_ft_sensor/force.x' does not exist`

ymollard, 2025-02-04:
Hi,

I am getting an error when simulating a force-torque sensor controller:
```
[gazebo-1] [ERROR] [1737905258.724732940] [controller_manager]: Caught exception of type : St13runtime_error while claiming the state interfaces.
Can't activate controller 'ft_sensor_controller': State interface with key 'wrist_ft_sensor/force.x' does not exist
[gazebo-1] [ERROR] [1737905258.724837112] [controller_manager]: Could not activate controller : 'ft_sensor_controller'
```
However the hardware interface seems good:
```
ros2 control list_hardware_interfaces | grep ft_sensor_controller
[INFO] [1737937074.447771896] [_ros2cli_170637]: waiting for service /controller_manager/list_hardware_interfaces to become available...
        ft_sensor_controller/wrist_ft_sensor/force.x
        ft_sensor_controller/wrist_ft_sensor/force.y
        ft_sensor_controller/wrist_ft_sensor/force.z
        ft_sensor_controller/wrist_ft_sensor/torque.x
        ft_sensor_controller/wrist_ft_sensor/torque.y
        ft_sensor_controller/wrist_ft_sensor/torque.z
```

This FT sensor controller did NOT raise error using ros2_controllers 4.18.0 and started successfully:
```
apt list ros-jazzy-ros2-controllers
ros-jazzy-ros2-controllers/noble,now 4.18.0-1noble.20241228.015625 amd64 [installed]
```
And now, since an `apt upgrade` made in end January, version 4.19.0 causes the force.x issue:
```
apt list ros-jazzy-ros2-controllers
ros-jazzy-ros2-controllers/noble,now 4.19.0-1noble.20250115.215226 amd64 [installed]
```

**To Reproduce**
So far I only know how to reproduce this issue by starting the Tiago Harmonic Gazebo simulation from https://github.com/Tiago-Harmonic/tiago_harmonic/ with the following launchfile:

```
ros2 launch tiago_gazebo tiago_gazebo.launch.py is_public_sim:=True world_name:=house
```

Maybe one could reproduce the issue by simulating a FT sensor controller alone in Gazebo using the state interface here above and ros2_controllers?

**Environment**
 - OS: Ubuntu 24.04
 - Version Jazzy

kavikode, 2025-02-06:
any updates please?

saikishor, 2025-02-06:
First, we have to check if the force torque interfaces are present without any controller on.

ymollard, 2025-02-07:
> First, we have to check if the force torque interfaces are present without any controller on.

Isn't that what I checked with `list_hardware_interfaces` here above? I tried to unload the controllers so that none is left but they refuse to unload since they are active. Anyway, the FT controller is unconfigured so that should not interfere with our interface checks?
```
mobile_base_controller  diff_drive_controller/DiffDriveController                     active      
arm_controller          joint_trajectory_controller/JointTrajectoryController         active      
torso_controller        joint_trajectory_controller/JointTrajectoryController         active      
joint_state_broadcaster joint_state_broadcaster/JointStateBroadcaster                 active      
ft_sensor_controller    force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster  unconfigured
head_controller         joint_trajectory_controller/JointTrajectoryController         active      
imu_sensor_broadcaster  imu_sensor_broadcaster/IMUSensorBroadcaster                   active      
gripper_controller      joint_trajectory_controller/JointTrajectoryController         active 
```

The controller is defined here under and this is Gazebo that is expected to expose this interface, so if `list_hardware_interfaces` does not help, I do not know what else should be checked :)

https://github.com/Tiago-Harmonic/tiago_robot/blob/jazzy/tiago_controller_configuration/config/ft_sensor_controller.yaml
```yaml
ft_sensor_controller:
  ros__parameters:
    type: force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster
    frame_id: wrist_ft_link
    sensor_name: wrist_ft_sensor
```

saikishor, 2025-02-07:
> Isn't that what I checked with list_hardware_interfaces here above? I tried to unload the controllers so that none is left but they refuse to unload since they are active. Anyway, the FT controller is unconfigured so that should not interfere with our interface checks?

The `list_hardware_interface` shows all the interfaces exported within the framework including the exported interfaces of the controllers

You cannot unload the controllers when they are active, you are supposed to deactivate them and then proceed with the unloading.

As you cannot find any force torque interfaces exposed by the Gazebo, then it should be something coming from ignition, but not from the ros2_control framework.

I hope this answers your question

ymollard, 2025-03-15:
Thanks for the answers @saikishor!

I'm still not able to dig that issue and understand what is going on with Tiago robot, but since according to you that is a bug from the controllers and not from the ros2_control framework, I'm closing that one.

closed 2025-03-15 by ymollard