ros-controls/ros2_controllers #1558: Parameters for force_torque_sensor_broadcaster

rcruzoliver, 2025-02-27:
**Environment (please complete the following information):**
 - OS: [Ubuntu 24 - Docker Container with base image Humble]
 - Version [Humble]

**Description**
force_torque_sensor_broadcaster parameters are not parsed properly and the controller does not start. The same launch file and config files work in jazzy.

This is the launch file

```
import os
import launch
import launch_ros.actions
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():
    # Get the package share directory
    package_share_directory = get_package_share_directory('bota_driver')

    # Locate the URDF file
    urdf_file = os.path.join(package_share_directory, 'urdf', 'bota_ft_sensor_standalone_robot.urdf')
    
    # Locate the ROS2 Control parameters file
    controller_manager_params = os.path.join(package_share_directory, 'config', 'controller_manager_params.yaml')

    # Locate the force-torque sensor broadcaster parameters file
    force_torque_sensor_broadcaster_params = os.path.join(package_share_directory, 'config', 'force_torque_sensor_broadcaster_params.yaml')

    # Read the URDF file contents
    with open(urdf_file, 'r') as file:
        robot_description = file.read()

    # Robot State Publisher
    robot_state_publisher_node = launch_ros.actions.Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        name='robot_state_publisher',
        output='screen',
        parameters=[{'robot_description': robot_description}]
    )

    # Controller Manager 
    controller_manager_node = launch_ros.actions.Node(
        package='controller_manager',
        executable='ros2_control_node',
        name='controller_manager',
        output='screen',
        parameters=[{'robot_description': robot_description}, controller_manager_params]
    )

    # Spawner for force-torque sensor broadcaster
    force_torque_sensor_broadcaster_spawner = launch_ros.actions.Node(
        package='controller_manager',
        executable='spawner',
        arguments=['force_torque_sensor_broadcaster', '--param-file', force_torque_sensor_broadcaster_params],
        output='screen',
    )

    return launch.LaunchDescription([
        robot_state_publisher_node,
        controller_manager_node,  # Start the controller manager properly
        force_torque_sensor_broadcaster_spawner  # Load the force sensor broadcaster
    ])`

```
This is controller_manager_params.yaml
```
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz
    use_sim_time: false
```  

This is the force_torque_sensor_broadcaster_params.yaml
```    
force_torque_sensor_broadcaster:
  ros__parameters:
    type: force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster
    frame_id: "sensor_frame"
    sensor_name: "bota_ft_sensor"
```  

I have also tried to pass all the parameters to the ros2_control_node...modifying the controller_manager_params.yaml to this:
```
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz
    use_sim_time: false

    force_torque_sensor_broadcaster:
      type: force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster
      frame_id: "sensor_frame"
      sensor_name: "bota_ft_sensor"   
```    
But it does also not work.

saikishor, 2025-02-27:
Hello @rcruzoliver 

Looking at the code, I've found these PRs in the past. Can you please verify if it is working with any of these : https://github.com/ros-controls/ros2_controllers/pull/698 (This has been reverted in https://github.com/ros-controls/ros2_controllers/pull/988)?. 

What I don't understand is why it is broken again now. Can you please also share the version of the ros2 control packages you have?. I'm mostly interested in `controller_manager`, `controller_interface` and `force_torque_sensor_broadcaster`.

Thank you

rcruzoliver, 2025-02-27:
controller manager = 2.48.0
hardware_interface = 2.48.0
controller_interface = 2.48.0
force_torque_sensor_broadcaster = 2.41.0

I am using the binaries, so as a first approach I have not tried the ideas in the PRs...if you confirm that the versions in the binaries are problematic, I will compile from source with the modifications.