ros-controls/ros2_controllers #1574: [CI] `test_force_torque_sensor_broadcaster` regularily times out

christophfroehlich, 2025-03-06:
```
test_results/force_torque_sensor_broadcaster/test_force_torque_sensor_broadcaster.gtest.xml
2025-03-06T02:46:41.8447062Z 2: [==========] Running 13 tests from 1 test suite.
2025-03-06T02:46:41.8447487Z 2: [----------] Global test environment set-up.
2025-03-06T02:46:41.8447818Z 2: [----------] 13 tests from ForceTorqueSensorBroadcasterTest
2025-03-06T02:46:41.8448266Z 2: [ RUN      ] ForceTorqueSensorBroadcasterTest.SensorName_InterfaceNames_NotSet
2025-03-06T02:46:41.8449075Z 2: [ERROR] [1741229141.866531381] [test_force_torque_sensor_broadcaster]: 'sensor_name' or at least one 'interface_names.[force|torque].[x|y|z]' parameter has to be specified.
2025-03-06T02:46:41.8449907Z 2: [       OK ] ForceTorqueSensorBroadcasterTest.SensorName_InterfaceNames_NotSet (12 ms)
2025-03-06T02:46:41.8450451Z 2: [ RUN      ] ForceTorqueSensorBroadcasterTest.SensorName_InterfaceNames_Set
2025-03-06T02:46:41.8451315Z 2: [ERROR] [1741229141.872119148] [test_force_torque_sensor_broadcaster]: both 'sensor_name' and 'interface_names.[force|torque].[x|y|z]' parameters can not be specified together.
2025-03-06T02:46:41.8452133Z 2: [       OK ] ForceTorqueSensorBroadcasterTest.SensorName_InterfaceNames_Set (5 ms)
2025-03-06T02:46:41.8452710Z 2: [ RUN      ] ForceTorqueSensorBroadcasterTest.SensorName_IsEmpty_InterfaceNames_NotSet
2025-03-06T02:46:41.8453534Z 2: [ERROR] [1741229141.876958103] [test_force_torque_sensor_broadcaster]: 'sensor_name' or at least one 'interface_names.[force|torque].[x|y|z]' parameter has to be specified.
2025-03-06T02:46:41.8454478Z 2: [       OK ] ForceTorqueSensorBroadcasterTest.SensorName_IsEmpty_InterfaceNames_NotSet (5 ms)
2025-03-06T02:46:41.8455083Z 2: [ RUN      ] ForceTorqueSensorBroadcasterTest.InterfaceNames_IsEmpty_SensorName_NotSet
2025-03-06T02:46:41.8455904Z 2: [ERROR] [1741229141.881561407] [test_force_torque_sensor_broadcaster]: 'sensor_name' or at least one 'interface_names.[force|torque].[x|y|z]' parameter has to be specified.
2025-03-06T02:46:41.8456757Z 2: [       OK ] ForceTorqueSensorBroadcasterTest.InterfaceNames_IsEmpty_SensorName_NotSet (5 ms)
2025-03-06T02:46:41.8457322Z 2: [ RUN      ] ForceTorqueSensorBroadcasterTest.SensorName_Configure_Success
2025-03-06T02:46:41.8457840Z 2: [       OK ] ForceTorqueSensorBroadcasterTest.SensorName_Configure_Success (6 ms)
2025-03-06T02:46:41.8458372Z 2: [ RUN      ] ForceTorqueSensorBroadcasterTest.InterfaceNames_Configure_Success
2025-03-06T02:46:41.8458863Z 2/2 Test #2: test_force_torque_sensor_broadcaster ........***Timeout  60.04 sec
```

Juliaj, 2025-03-11:
@christophfroehlich , I'm interested in looking into this, a good opportunity for me to learn another controller.

 Just for my knowledge, the `**Timeout  60.04 sec` in the note above was a timeout from running entire test suite or was it a result from a single long running test?

christophfroehlich, 2025-03-11:
This is from tests in the CI.
normally the test runs only for a couple of seconds, but sometimes it gets stuck and runs in the timeout.

christophfroehlich, 2025-03-15:
This PID controller issue looks similar (jazzy)
https://github.com/ros-controls/ros2_control_ci/actions/runs/13868494155/job/38811899016

christophfroehlich, 2025-03-25:
See https://github.com/ros-controls/ros2_controllers/issues/1608 with a similar issue

christophfroehlich, 2025-04-04:
Still happening, but https://github.com/ros-controls/ros2_control/pull/2142 is not released yet
https://github.com/ros-controls/ros2_controllers/actions/runs/14256953898/job/39961242259

christophfroehlich, 2025-04-05:
https://github.com/ros-controls/ros2_controllers/issues/1628

christophfroehlich, 2025-04-06:
Still fails on jazzy branch :/ https://github.com/ros-controls/ros2_controllers/issues/1629

saikishor, 2025-04-06:
@Juliaj I've checked the code and I think it might be caused by lock of RTPublisher : (https://github.com/ros-controls/ros2_controllers/blob/master/force_torque_sensor_broadcaster%2Fsrc%2Fforce_torque_sensor_broadcaster.cpp#L105)

It would be great to add some log info prints both before and after this as well as in the tests with configure, activate and deactivate. This might give us some clue on where the issue might be coming from

saikishor, 2025-04-06:
> Do we have to release that? Or do we hope that it will happen on any source build?

Let's hope that it happens on source build

saikishor, 2025-04-13:
> @Juliaj I've checked the code and I think it might be caused by lock of RTPublisher : [...]

@Juliaj do you want to take care of it?

Juliaj, 2025-04-15:
> do you want to take care of it?

will do shortly.

christophfroehlich, 2025-04-16:
> > Do we have to release that? Or do we hope that it will happen on any source build?
> 
> Let's hope that it happens on source build

@saikishor Ah, you wanted logging in the controller instead of inside RTpublisher. Then we'll see this in other builds as well. 
However,  https://github.com/ros-controls/ros2_control_ci/pull/299 is needed to fix the source builds

saikishor, 2025-04-16:
Thanks for bringing it up

christophfroehlich, 2025-04-30:
```
2025-04-29T02:47:30.4774444Z 2: [ RUN      ] ForceTorqueSensorBroadcasterTest.InterfaceNames_Configure_Success
2025-04-29T02:47:30.4775637Z 2: [INFO] [1745894789.446491128] [test_force_torque_sensor_broadcaster]: FTSBroadcaster setup done
2025-04-29T02:47:30.4776524Z 2: [INFO] [1745894789.446515884] [test_force_torque_sensor_broadcaster]: Setting up interface names
2025-04-29T02:47:30.4777599Z 2: [INFO] [1745894789.446578952] [test_force_torque_sensor_broadcaster]: Setting up frame_id
2025-04-29T02:47:30.4778536Z 2: [INFO] [1745894789.446629186] [test_force_torque_sensor_broadcaster]: Calling on_configure
2025-04-29T02:47:30.4779459Z 2: [INFO] [1745894789.446661145] [test_force_torque_sensor_broadcaster]: Initialized force_torque_sensor with interface names fts_sensor/force.x, , , , , fts_sensor/torque.z
2025-04-29T02:47:30.4780246Z 2: [INFO] [1745894789.447014505] [test_force_torque_sensor_broadcaster]: Locking realtime publisher
2025-04-29T02:47:30.4780779Z 2: [INFO] [1745894789.447038730] [test_force_torque_sensor_broadcaster]: Locked realtime publisher
2025-04-29T02:47:30.4781326Z 2: [INFO] [1745894789.447046575] [test_force_torque_sensor_broadcaster]: Unlocking realtime publisher
2025-04-29T02:47:30.4781908Z 2: [INFO] [1745894789.447057495] [test_force_torque_sensor_broadcaster]: Unlocked realtime publisher
2025-04-29T02:47:30.4782632Z 2: [INFO] [1745894789.447064388] [test_force_torque_sensor_broadcaster]: Configure successful
2025-04-29T02:47:30.4783105Z 2: [INFO] [1745894789.447072844] [test_force_torque_sensor_broadcaster]: In TearDown
2025-04-29T02:47:30.4783553Z 2/2 Test #2: test_force_torque_sensor_broadcaster ........***Timeout  60.03 sec
```
we have the first timeout since the additional outputs were merged. Does this help?
https://github.com/ros-controls/ros2_controllers/issues/1665

saikishor, 2025-04-30:
This is very interesting. Thank you @christophfroehlich @Juliaj  :)

closed 2025-05-02 by christophfroehlich