ros-controls/ros2_controllers #1577: Implement a generic `filters` plugin

christophfroehlich, 2025-03-09:
The [filters](https://wiki.ros.org/filters) package provides a standardized interface for processing data as a sequence of filters. We also have custom filter plugins in the [control_toolbox](https://github.com/ros-controls/control_toolbox/tree/ros2-master/include/control_filters) package.

There exists already a blueprint of such a controller from the last years ROSCon workshop:

- https://github.com/ros-controls/roscon2024_control_workshop/blob/master/README.md#task-1-controller-chaining
- https://github.com/ros-controls/roscon2024_control_workshop/blob/master/workshop_controllers/src/chained_filter_done.cpp

Necessary steps:

- [ ] Move that to this repository
- [ ] spice it up with some tests (have a look at other controllers here)
- [ ] Advertise the control_filters of [control_toolbox](https://github.com/ros-controls/control_toolbox/tree/ros2-master/include/control_filters) in the description
- [ ] Create a follow-up PR in the demos repository to use such a filter for state interfaces of a controller chain.

CodrAyush, 2025-03-13:
@christophfroehlich  i want to work over this issue..could you pls help me out in setting up this project

ankurbodhe, 2025-03-17:
@christophfroehlich i'm interested to take this issue up and contribute.

christophfroehlich, 2025-03-18:
> @christophfroehlich i'm interested to take this issue up and contribute.

Great! Come back to me if you need advise or intermediate review!

kscottz, 2025-04-02:
@ankurbodhe how is it going? I haven't see a pull request. Did I miss it?

ankurbodhe, 2025-04-02:
I'm working on it and would have a PR up by 04/08.... let me know this is needed earlier than that

kscottz, 2025-04-08:
@ankurbodhe friendly ping.