ros-controls/ros2_controllers #1590: pid_controller: add reference interfaces for gains

christophfroehlich, 2025-03-12:
Currently, the pid_controller can be dynamically tuned by changing parameters.

Proposal: It would be useful for gain-scheduling use cases to export the gains as reference interfaces, which can then be claimed by preceding controllers.

rehanshah17, 2025-03-21:
Could I take up this issue please?

rehanshah17, 2025-03-23:
I think I was able to figure out how to export the pid constants as reference interfaces. Could you lead me down the right path of how the controllers would be able to claim this?

christophfroehlich, 2025-03-23:
let us get #1601 working first.

christophfroehlich, 2025-11-20:
Imho P,I,D gains will be sufficient for the beginning. Extending it with limits etc will be an easier step later in case someone has a usecase for it.

In chained mode, the controller manager should take care about the order of the controllers and call the update of the preceding controller first, i.e., in chained mode there should not be a need for "default values".

kumar-sanjeeev, 2025-11-21:
> Imho P,I,D gains will be sufficient for the beginning. Extending it with limits etc will be an easier step later in case someone has a usecase for it.
>
> In chained mode, the controller manager should [...]

understood and thanks for clarification.