ros-controls/ros2_controllers #1591: mecanum_drive_controller does not generate an odom frame ROS2 Jazzy

Shadow-Haund, 2025-03-14:
I have a robot with mecanum drive, and I don't know why, but it did not generate an odom frame after successful loading. The thing is that controller is loaded and I can control my robot via topic commands
This is my .yaml file:
`controller_manager:
  ros__parameters:
    update_rate: 50  # Hz

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    controller:
      type: mecanum_drive_controller/MecanumDriveController

joint_state_broadcaster:
  ros__parameters:
    type: "joint_state_broadcaster/JointStateBroadcaster"
    update_rate: 50

controller:
  ros__parameters:
    type: "mecanum_drive_controller/MecanumDriveController"
    reference_timeout: 0.9
    update_rate: 50

    front_left_wheel_command_joint_name: "front_left_wheel_joint"
    front_right_wheel_command_joint_name: "front_right_wheel_joint"
    rear_right_wheel_command_joint_name: "rear_right_wheel_joint"
    rear_left_wheel_command_joint_name: "rear_left_wheel_joint"
    enable_odom_tf: true

    kinematics:
      base_frame_offset: { x: 0.0, y: 0.0125, theta: 0.0 }
      wheels_radius: 0.0305
      sum_of_robot_center_projection_on_X_Y_axis: 0.182

    base_frame_id: "world_link"
    odom_frame_id: "odom"
    twist_covariance_diagonal: [0.0, 7.0, 14.0, 21.0, 28.0, 35.0]
    pose_covariance_diagonal: [0.0, 7.0, 14.0, 21.0, 28.0, 35.0]
`
And here is a tf tree

[image]

I'm also update all my ros2 packages, including controller, before creating this issue.
Appreciate any help in clarification of this problem.
Sorry for any mistakes in the issue description, English is not my native language. Thank you in advance.

Shadow-Haund, 2025-03-21:
Thank you for your answer, I'll check it. 
Sorry about inconvenience, I'll go to https://robotics.stackexchange.com in case there will be any new such questions from me.

PhilerXu, 2026-01-15:
> Without seeing your urdf files and knowing your use case, my guess is that `base_frame_id: "world_link" ` needs to be changed to `base_frame_id: "base_link"`. I have a [working setup](https://github.com/Juliaj/ros2_control_demos/tree/async_controller/example_17) involving mecanum_drive_controller if you need a reference.
> 
> By the way, https://robotics.stackexchange.com/ is a better place for questions like yours, since this git repository is generally for development discussions (features, issue reporting, etc.).

I am encountering the same issue.  Is it resolved? @Juliaj Which ROS version should I use to compile your code?