ros-controls/ros2_controllers #1595: [ERROR] [ros2_control_node-9]: process has died [pid 93435, exit code -6

mark5351-sudo, 2025-03-17:
I encountered this issue while initiating the robot; however, the hardware interface functions correctly when executed independently.
I'm currently use ros2 jazzy 

[image]
[image]

mark5351-sudo, 2025-03-17:
I am using these versions
ros2 pkg xml -t version ros2_control
4.27.0
ros2 pkg xml -t version ros2_controllers
4.21.0

christophfroehlich, 2025-03-17:
See the error message
`[ros2_control_node-9]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /home/rare/test_ws/install/minibot/share/minibot/config/controller.yaml'. Error: Error parsing a event near line 56, at ./src/parse.c:782, at ./src/rcl/arguments.c:415`

can you confirm that the path of the yaml file exists, and that it has correct syntax?

mark5351-sudo, 2025-03-17:
The YAML path exists, however, I require clarification on the syntax.

controller_manager:
  ros__parameters:
    update_rate: 30  # Hz
    # use_sim_time: true
    
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

diff_drive_controller:
  ros__parameters:
    type: diff_drive_controller/DiffDriveController
    left_wheel_names: ["left_wheel_joint"]
    right_wheel_names: ["right_wheel_joint"]

    wheel_separation: 0.266
    # wheels_per_side: 1  # actually 2, but both are controlled by 1 signal
    wheel_radius: 0.034

    wheel_separation_multiplier: 1.0
    left_wheel_radius_multiplier: 1.0
    right_wheel_radius_multiplier: 1.0

    publish_rate: 30.0
    odom_frame_id: odom
    base_frame_id: base_link
    pose_covariance_diagonal : [0.001, 0.001, 0.001, 0.001, 0.001, 0.01]
    twist_covariance_diagonal: [0.001, 0.001, 0.001, 0.001, 0.001, 0.01]

    open_loop: true
    enable_odom_tf: true

    cmd_vel_timeout: 0.5
    # publish_limited_velocity: true
    use_stamped_vel: false
    # velocity_rolling_window_size: 10

    # Velocity and acceleration limits
    # Whenever a min_* is unspecified, default to -max_*
    linear.x.has_velocity_limits: true
    linear.x.has_acceleration_limits: true
    linear.x.has_jerk_limits: false
    linear.x.max_velocity: 1.0
    linear.x.min_velocity: -1.0
    linear.x.max_acceleration: 1.0
    linear.x.max_jerk: 0.0
    linear.x.min_jerk: 0.0

    angular.z.has_velocity_limits: true
    angular.z.has_acceleration_limits: true
    angular.z.has_jerk_limits: false
    angular.z.max_velocity: 1.0
    angular.z.min_velocity: -1.0
    angular.z.max_acceleration: 1.0
    angular.z.min_acceleration: -1.0
    angular.z.max_jerk: 0.0
    angular.z.min_jerk: 0.0

christophfroehlich, 2025-03-17:
Line 56 is the last line, right? can you check if you have some special characters there (whitespace etc), add a newline at the end, or just delete the linear/angular limits for a test?

mark5351-sudo, 2025-03-18:
The previous error on line 56 has been resolved; however, the [ERROR] [ros2_control_node-9] issue persists.
```
ros2 launch minibot robot.launch.py
[INFO] [launch]: All log files can be found below /home/rare/.ros/log/2025-03-18-11-42-10-444404-wartier-17386
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [17390]
[INFO] [twist_mux-2]: process started with pid [17391]
[INFO] [twist_stamper-3]: process started with pid [17392]
[INFO] [republish-4]: process started with pid [17393]
[INFO] [republish-5]: process started with pid [17394]
[INFO] [republish-6]: process started with pid [17395]
[INFO] [republish-7]: process started with pid [17396]
[INFO] [sllidar_node-8]: process started with pid [17397]
[INFO] [ros2_control_node-9]: process started with pid [17422]
[INFO] [spawner-10]: process started with pid [17453]
[INFO] [spawner-11]: process started with pid [17465]
[republish-7] [INFO] [1742272930.801549092] [image_transport_republish_zstd_image_raw]: The 'in_transport' parameter is set to: raw
[republish-7] [INFO] [1742272930.801683780] [image_transport_republish_zstd_image_raw]: The 'out_transport' parameter is set to: 
[republish-4] [INFO] [1742272930.810562324] [image_transport_republish_compressed_image_raw]: The 'in_transport' parameter is set to: raw
[republish-4] [INFO] [1742272930.810713630] [image_transport_republish_compressed_image_raw]: The 'out_transport' parameter is set to: 
[republish-5] [INFO] [1742272930.811054394] [image_transport_republish_compressedDepth_image_raw]: The 'in_transport' parameter is set to: raw
[republish-5] [INFO] [1742272930.811352625] [image_transport_republish_compressedDepth_image_raw]: The 'out_transport' parameter is set to: 
[republish-6] [INFO] [1742272930.822338742] [image_transport_republish_theora_image_raw]: The 'in_transport' parameter is set to: raw
[republish-6] [INFO] [1742272930.822882473] [image_transport_republish_theora_image_raw]: The 'out_transport' parameter is set to: 
[robot_state_publisher-1] [INFO] [1742272930.831141188] [robot_state_publisher]: Robot initialized
[twist_mux-2] [INFO] [1742272930.836827472] [twist_mux]: Topic handler 'topics.joystick' subscribed to topic 'joy_vel': timeout = 0.500000s , priority = 100.
[twist_mux-2] [INFO] [1742272930.851454851] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'nav_vel': timeout = 0.500000s , priority = 10.
[sllidar_node-8] [INFO] [1742272930.857599934] [sllidar_node]: SLLidar running on ROS2 package SLLidar.ROS2 SDK Version:1.0.1, SLLIDAR SDK Version:2.1.0
[sllidar_node-8] [INFO] [1742272930.866099146] [sllidar_node]: SLLidar S/N: DEBC9A87C5E392D5A5E492F84E69316B
[sllidar_node-8] [INFO] [1742272930.866154792] [sllidar_node]: Firmware Ver: 1.27
[sllidar_node-8] [INFO] [1742272930.866166988] [sllidar_node]: Hardware Rev: 6
[sllidar_node-8] [INFO] [1742272930.867199847] [sllidar_node]: SLLidar health status : 0
[sllidar_node-8] [INFO] [1742272930.867238498] [sllidar_node]: SLLidar health status : OK.
[ros2_control_node-9] [INFO] [1742272930.901818955] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-9] [INFO] [1742272930.939960238] [controller_manager]: update rate is 30 Hz
[ros2_control_node-9] [INFO] [1742272930.940021381] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-9] [INFO] [1742272930.940162033] [controller_manager]: Successful set up FIFO RT scheduling policy with priority 50.
[ros2_control_node-9] [INFO] [1742272930.940744329] [controller_manager]: Received robot description from topic.
[ros2_control_node-9] [INFO] [1742272930.950026923] [controller_manager.resource_manager]: Loading hardware 'RobotSystem' 
[ros2_control_node-9] [INFO] [1742272930.951815638] [controller_manager.resource_manager]: Loaded hardware 'RobotSystem' from plugin 'diffdrive_arduino/DiffDriveArduinoHardware'
[ros2_control_node-9] [INFO] [1742272930.951902506] [controller_manager.resource_manager]: Initialize hardware 'RobotSystem' 
[ros2_control_node-9] [INFO] [1742272930.952419382] [DiffDriveArduinoHardware]: PID values not supplied, using defaults.
[ros2_control_node-9] [INFO] [1742272930.952449527] [controller_manager.resource_manager]: Successful initialization of hardware 'RobotSystem'
[ros2_control_node-9] [INFO] [1742272930.952867950] [resource_manager]: 'configure' hardware 'RobotSystem' 
[ros2_control_node-9] [INFO] [1742272930.952880598] [DiffDriveArduinoHardware]: Configuring ...please wait...
[ros2_control_node-9] [ERROR] [1742272930.952968476] [controller_manager.resource_manager]: Exception of type : N9LibSerial10OpenFailedE occurred while configuring hardware 'RobotSystem': Bad file descriptor
[ros2_control_node-9] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-9]   what():  Failed to set the initial state of the component : RobotSystem to active
[ros2_control_node-9] Stack trace (most recent call last):
[ros2_control_node-9] #18   Object "/usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2", at 0xffffffffffffffff, in 
[sllidar_node-8] [INFO] [1742272931.069214544] [sllidar_node]: current scan mode: Sensitivity, sample rate: 16 Khz, max_distance: 25.0 m, scan frequency:10.0 Hz, 
[spawner-11] [INFO] [1742272931.434004917] [spawner_diff_drive_controller]: waiting for service /controller_manager/list_controllers to become available...
[... cut ...]
publishing #3: geometry_msgs.msg.TwistStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id='base_link'), twist=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=1.5, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=1.0, z=0.5)))

publishing #4: geometry_msgs.msg.TwistStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id='base_link'), twist=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=1.5, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=1.0, z=0.5)))
```

christophfroehlich, 2025-03-18:
Read carefully, there is a different error now
```
[ros2_control_node-9] [ERROR] [1742272930.952968476] [controller_manager.resource_manager]: Exception of type : N9LibSerial10OpenFailedE occurred while configuring hardware 'RobotSystem': Bad file descriptor
[ros2_control_node-9] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-9]   what():  Failed to set the initial state of the component : RobotSystem to active
```

The thrown error comes from LibSerial, this is not related to ros2_control any more.

What is the difference between the two launch files?

christophfroehlich, 2025-03-18:
> The previous error on line 56 has been resolved; 

Could you please tell us what was the problem with your yaml file before?

closed 2025-03-18 by mark5351-sudo