ros-controls/ros2_controllers #1604: Add Model Predictive Control (MPC) Controller to ros2_control

yassine-cherni, 2025-03-23:
I’ve been working with ros2_control for my 4-wheel robot, and I love how easy it is to use controllers like DiffDriveController or MecanumDriveController. But I often get frustrated because their PID-based approach can’t handle things like velocity limits, complex dynamics, or balancing multiple goals—like following a path while avoiding obstacles. It feels limiting when I want more advanced control at the low level.

I’d love to see a new MPCController in ros2_controllers that works seamlessly with ros2_control. It would use Model Predictive Control (MPC) to predict the robot’s behavior, optimize control inputs, and respect constraints like velocity or acceleration limits. I imagine configuring it with a YAML file to set the robot model, constraints, and goals, using a library like CasADi to make it efficient for robots like mine.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

christophfroehlich, 2025-03-24:
A while ago I made a proposal for custom controller plugins for the JoinTrajectoryController: #885
I use this for exactly this reason, a model-predictive controller for a robot arm with passive joints.

I'm not sure how a generalized approach could look like, but the JTC might be a good start to have a reasonable set of use cases.

closed 2025-03-26 by yassine-cherni

christophfroehlich, 2025-03-26:
I didn't want to stop you from making a proposal ;) I never worked with CasADi so far, but I'm eager to learn. I suggest that you describe a bit the problem you want to solve and how you would attack it. As I said, it might be too much for a general purpose MPC controller for all kind of robots.