ros-controls/ros2_controllers #1613: Tricycle Controller: Traction Joint and Steering Joint

theinge9, 2025-03-28:
Hello,

I am using Tricycle Controller to test with my tricycle-wheeled robot in simulation and I noticed that the traction wheels are moving when the robot tends to rotate. Is there any options and parameters to adjust for controlling steering joint without connecting with driving wheels.For instances, wheel rotation is stopped when the steering is rotating.

Always appreciate to your amazing works.
Best Regards
ge

christophfroehlich, 2025-03-28:
Do you mean tricycle_controller or tricycle_steering_controller?

- tricycle_controller: a single actuated wheel, with traction and steering
- tricycle_steering_controller: single steering wheel, traction wheel on a different axle.

Both have a mechanism to stop traction if the steering takes more time (see `reduce_wheel_speed_until_steering_reached` parameter of tricycle_steering_controller). Is this what you are looking for?

theinge9, 2025-03-28:
My robot structure is only front wheel traction and steering, so i used it with tricycle_controller and is there reduce_wheel_speed_until_steering_reached parameter in tricycle_controller?

The question that i want to know is when the robot received angular_z value only with linear_x:0, the robot rotates a little with the front movement.So is there any control timing for steering only without moving the robot.

Thanks for your support to my issue.

christophfroehlich, 2025-03-28:
no, the behavior of reduce_wheel_speed_until_steering_reached is always active in tricycle controller. But it does not wait until fully steered, there is some continuous function implemented.

https://github.com/ros-controls/ros2_controllers/blob/996f0304cbfef89f6162442e4aa7aefd3ed4d843/tricycle_controller/src/tricycle_controller.cpp#L167-L183

The spin action is implemented as a special case here
https://github.com/ros-controls/ros2_controllers/blob/996f0304cbfef89f6162442e4aa7aefd3ed4d843/tricycle_controller/src/tricycle_controller.cpp#L539-L544

So I think the problem is that it will move a bit until it is fully steered, right?

theinge9, 2025-03-28:
Yes, and there is no control for the steering in desired angle rotation.Currently, the robot just rotate with fully steering and return automatically to home position.

christophfroehlich, 2025-03-28:
So what is the problem actually?

theinge9, 2025-04-01:
I just want to sure about the control behavior of Tricycle Controller.
Thank you for your explanation.

closed 2025-04-01 by theinge9