ros-controls/ros2_controllers #1643: [JSB] how to fix joints order

RainJCloude, 2025-04-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. Foxy]
- Anything that may be unusual about your environment

**Additional context**
Add any other context about the problem here, especially include any modifications to any ros2_controllers that relate to this issue.

christophfroehlich, 2025-04-18:
On Jazzy+Rolling https://github.com/ros-controls/ros2_controllers/pull/1572

closed 2025-04-18 by christophfroehlich

RainJCloude, 2025-04-18:
Sorry, my issue was mainly related on how to retrieve that commit. I have installed the package with apt-get so I dont know how and if I can switch between branches. Moreover, I would also like to know if there is possibility to fix this on humble, despite I have a container with jazzy and I could transfer my project there