ros-controls/ros2_controllers #1644: [JSB] How to retrieve the commit which fix the joints-order

RainJCloude, 2025-04-18:
My previous issue was closed immediately, so I’m reopening it.

I am experiencing the problem described in #1572, and I would like to ask:

- How can I retrieve that commit, since I have installed the package with:    `  sudo apt-get install ros-humble-ros2-controllers  ` 
- There is some chance to fix it also in humble

christophfroehlich, 2025-04-18:
You haven't added any information to the issue template, that's why I closed it ;)

You can compile the latest rolling version on humble distro, see "Development Version" [here](https://control.ros.org/humble/doc/getting_started/getting_started.html#building-from-source).

christophfroehlich, 2025-04-18:
We try to backport it, keep an eye on
https://github.com/ros-controls/ros2_controllers/pull/1635

Once it is merged, you can just compile this single package without the need for compiling the whole stack like above.

closed 2025-04-19 by RainJCloude