ros-controls/ros2_controllers #1660: Call lifecycle transitions instead of callbacks in tests

christophfroehlich, 2025-04-26:
## Background

In ros2_control we implement managed lifecycle nodes similar to https://design.ros2.org/articles/node_lifecycle.html
The `on_xx()` methods are transition callbacks, which are registered to the transitions in the controller_interface_base class [here](https://github.com/ros-controls/ros2_control/blob/cf19faf3464c67918b69e7d53a01a237e5a7b043/controller_interface/src/controller_interface_base.cpp#L80-L129).

The controller in the tests should follow the lifecycles, but we already have a mixture of using callback only
https://github.com/ros-controls/ros2_controllers/blob/496e39dc53caaaea4795b26f0ba5fa8c0e550877/diff_drive_controller/test/test_diff_drive_controller.cpp#L435
and using the correct transitions
https://github.com/ros-controls/ros2_controllers/blob/496e39dc53caaaea4795b26f0ba5fa8c0e550877/joint_trajectory_controller/test/test_trajectory_controller_utils.hpp#L415

The issue here is that the transition callbacks are used instead of triggering the lifecylce node callbacks, i.e.,
changing occurrences of
```
controller->on_configure()
controller->on_activate()
```
to 

```
controller->configure()
controller->get_node()->activate()
```
should do the job (compare with other controller tests).

Slightly related: I addressed the configure() call with https://github.com/ros-controls/ros2_controllers/pull/1659

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

LalitNarayanYadav, 2025-04-27:
Hi! I would like to claim this issue and start working on it. I’ll focus on replacing the transition callbacks with the lifecycle node transitions as described. 

Looking forward to contributing and will reach out if I encounter any questions. Thanks for the opportunity!

christophfroehlich, 2025-04-27:
@LalitNarayanYadav let's focus on one topic, I advise that you first work on the CMakeLists topic ;)

jsantoso91, 2025-05-13:
Hi @christophfroehlich, do you mind assigning this issue to me? I am planning to submit the follow up fix for #1658 using this ticket as well. 

Also, should I try to address similar pattern in other tests as well? I am thinking to create a draft pull request for test_diff_drive_controller.cpp and the other two tests in #1658 first, then get some feedback before trying to update the remaining tests (if there are any).

Thanks!