ros-controls/ros2_controllers #1661: Modernize CMakeLists

christophfroehlich, 2025-04-26:
## Background
On rolling, we get such compile-time warnings:
```
   CMake Deprecation Warning at /opt/ros/rolling/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
    ament_target_dependencies() is deprecated.  Use target_link_libraries()
    with modern CMake targets instead.  Try replacing this call with:
  
      target_link_libraries(diff_drive_controller PUBLIC
        ${geometry_msgs_TARGETS}
        ${nav_msgs_TARGETS}
        ${tf2_msgs_TARGETS}
        control_toolbox::control_toolbox
        control_toolbox::exponential_filter
        control_toolbox::exponential_filter_parameters
        control_toolbox::gravity_compensation
        control_toolbox::gravity_compensation_filter_parameters
        control_toolbox::low_pass_filter
        control_toolbox::low_pass_filter_parameters
        control_toolbox::rate_limiter
        control_toolbox::rate_limiter_parameters
        controller_interface::controller_interface
        hardware_interface::hardware_interface
        hardware_interface::mock_components
        pluginlib::pluginlib
        rclcpp::rclcpp
        rclcpp_lifecycle::rclcpp_lifecycle
        rcpputils::rcpputils
        realtime_tools::realtime_tools
        realtime_tools::thread_priority
        tf2::tf2
      )
```
Try to do what is proposed here.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

LalitNarayanYadav, 2025-04-27:
Hi, I’d like to work on this issue. I will follow the instructions and update the CMakeLists to replace the deprecated `ament_target_dependencies()` call with `target_link_libraries()` as suggested. Thanks!

christophfroehlich, 2025-04-27:
fyi: This applies to all packages in ros-controls, especially too in
https://github.com/ros-controls/ros2_control/issues/2207
https://github.com/ros-controls/ros2_control_demos/issues/790