ros-controls/ros2_controllers #1664: [Effort controller] Set initial config

RainJCloude, 2025-04-27:
Hello, I would like to use effort controller to command robots my I don't manage to give it an initial config.
I can do that just if I set a position controller setting the joints value in the state interface, but these values are ignored if I used an effort controller, or if I still use a position controller with both position and effort command interfaces.

Do you have any suggestions on how to cope with this problem?
Thanks a lot

saikishor, 2025-04-27:
@RainJCloude do you mind explaining what you mean by these values being ignored by the effort controller?

RainJCloude, 2025-04-27:
```
    <xacro:macro name="joint_ros2_control" params="name initial_pos">
        <joint name="${name}">
            <command_interface name="position">
                <param name="min">-3.14</param>
                <param name="max">3.14</param>
              </command_interface>
            <state_interface name="position"> 
                <param name="initial_value">${initial_pos}</param>
            </state_interface>
            <state_interface name="velocity"/>
            <state_interface name="effort"/>
        </joint> 
    </xacro:macro>
```

That here in ```<state_interface name="position"> ``` i have given the initial configuraton. If I load a position controller the robot effectively start with that initial configuration, but if I use an effort controller, it doesnt. I wonder how could I set initial config when I need effort controllers

christophfroehlich, 2025-04-27:
Setting initial states from the ros2_control tag is simulator-specific and not handled by ros2_control itself. 
Are you using Gazebo classic, or modern gazebo?

RainJCloude, 2025-04-27:
Oh, I see. I am using modern gazebo (harmonic version). So I should make an issue in the gz-sim repository?

saikishor, 2025-04-27:
> Oh, I see. I am using modern gazebo (harmonic version). So I should make an issue in the gz-sim repository?
> The fact is, iit sounded strange to me that when using both the position and effort command interfaces, Gazebo ignored the initial configuration, whereas when using just one, it spawned in the correct pose

Yes, please open it in gz_ros2_control repository