ros-controls/ros2_controllers #1676: Update mecanum_drive_controller with common controller features

KmakD, 2025-05-09:
It appears that the mecanum_drive_controller is lacking some features that are present in other controllers and seem to be important or useful. Compared to the diff_drive_controller, there are several features that could be implemented in the mecanum_drive_controller:

- Velocity limits (`linear.x`, `linear.y`, `angular.z`)
- `cmd_vel_timeout`: Timeout in seconds, after which input command on cmd_vel topic is considered staled.
- tf frame prefix parameters:
  - `tf_frame_prefix_enable`:  Enables or disables appending tf_prefix to tf frame id's.
  - `tf_frame_prefix`: (optional) Prefix to be appended to the tf frames, will be added to odom_id and base_frame_id before publishing. If the parameter is empty, controller's namespace will be used.
- Wheel radius multiplier - Correction factor when radius of wheel differs from the nominal value in `wheel_radius` parameter.
- `publish_rate`: Publishing rate (Hz) of the odometry and TF messages.

All of these features are already implemented in the diff_drive_controller, so it seems the task would be to port them into the mecanum_drive_controller.

christophfroehlich, 2025-05-10:
regarding the velocity limits, I planned to create a chainable TwistLimiter which can be used with all mobile robot controllers, instead of duplicating the code in all of them.

Regarding the other points, contributions are welcome!

KmakD, 2025-05-12:
> wheel radius: why one would need that, instead of changing the original parameter?
It's a thing in diff drive controller also separated to left and right wheels. I guess it makes sense there because you can have inflated tires and you may want to correct nominal radius value depending on pressure in the wheels. For mecanum, it looks like it would be useless.

Do you have any idea, when can we expect that TwistLimiter?

KmakD, 2025-05-14:
I figured that `cmd_vel_timeout` in mecanum coresponds to `reference_timeout`. This made me notice another inconsistency between different controllers regarding topics. For diff and tricycle topic are:
`~/cmd_vel`
`~/odom`
`/tf`

But for mecanum and controllers that use steering_controllers_library it is:
`~/reference`
`~/odometry`
`~/tf_odometry`

Do you plan to unify that at some point? Personally, I find `~/tf_odometry` a bit strange, knowing that some standard tools like Rviz require a global `/tf` topic. In most cases this topic would need to be remapped to be usefull anyway.

Juliaj, 2025-06-10:
> Do you plan to unify that at some point? Personally, I find `~/tf_odometry` a bit strange, knowing that some standard tools like Rviz require a global `/tf` topic. In most cases this topic would need to be remapped to be usefull anyway.

+1...  This inconsistency can be mentally taxing when working with these controllers, as it requires additional effort to manage and adapt to different naming conventions. I'd very much like to see a consolidation for a more consistent naming.

destogl, 2025-07-02:
We shoudl adjust `DiffDriveController` to the new namings. There is a very old naming of topics used there.

christophfroehlich, 2025-07-30:
Regarding TwistLimiter: https://github.com/ros-controls/ros2_controllers/pull/1801#issuecomment-3137450292

christophfroehlich, 2025-11-01:
Topic renaming and harmonizing the parameters is still missing. Anyone wants to tackle this?

We need to do this gracefully, i.e., don't remove but deprecate usage of the old ones. For parameters it is easy to deprecate the old one, but publishers need to use `Node::count_subscribers()` to throw a warning when a node subscribes to it.

I think 
```
~/cmd_vel
```
is widely adopted, for tf+odometry I think everything is valid because any consumer has to be configured anyways. Wheel odometry republished as tf2 is still only wheel odometry and will drift without robot localization. Therefore,
```
~/odometry
~/tf_odometry
```
seems to be fine as well.