ros-controls/ros2_controllers #1681: Problem with mecanum_drive_controller control failure in ROS 2 Humble and Gazebo simulation.

wahaha-1, 2025-05-14:
I have added the ros2_control plugin to my robot and completed its definition, and it launched successfully.
After adding and defining the ros2_control plugin for my robot and successfully launching it, I found that controlling it with
` ros2 run teleop_twist_keyboard teleop_twist_keyboard `
only allows forward/backward movement. Furthermore, occasionally the robot spins upwards uncontrollably like helicopter blades before dropping down.
I would like to know how to control the robot to move correctly. Please find the relevant files in the attachment. Awaiting your response.

[jetauto_ros2_controller.txt](https://github.com/user-attachments/files/20208533/jetauto_ros2_controller.txt)

[jetauto_ros2_controller.txt](https://github.com/user-attachments/files/20208538/jetauto_ros2_controller.txt)

[ros2_control.txt](https://github.com/user-attachments/files/20208564/ros2_control.txt)

[jetauto_car.urdf.txt](https://github.com/user-attachments/files/20208581/jetauto_car.urdf.txt)

```
[gzserver-1] [INFO] [1747229499.563436332] [resource_manager]: Successful initialization of hardware 'jetauto_hardware_interface'
[gzserver-1] [INFO] [1747229499.563506974] [resource_manager]: 'configure' hardware 'jetauto_hardware_interface' 
[gzserver-1] [INFO] [1747229499.563513938] [resource_manager]: Successful 'configure' of hardware 'jetauto_hardware_interface'
[gzserver-1] [INFO] [1747229499.563523789] [resource_manager]: 'activate' hardware 'jetauto_hardware_interface' 
[gzserver-1] [INFO] [1747229499.563528141] [resource_manager]: Successful 'activate' of hardware 'jetauto_hardware_interface'
[gzserver-1] [INFO] [1747229499.563579062] [gazebo_ros2_control]: Loading controller_manager
[gzserver-1] [WARN] [1747229499.571637554] [gazebo_ros2_control]:  Desired controller update period (0.02 s) is slower than the gazebo simulation period (0.001 s).
[gzserver-1] [INFO] [1747229499.571772626] [gazebo_ros2_control]: Loaded gazebo_ros2_control.
[INFO] [spawn_entity.py-4]: process has finished cleanly [pid 9832]
[INFO] [spawner-6]: process started with pid [10023]
[INFO] [spawner-7]: process started with pid [10025]
[gzserver-1] [INFO] [1747229499.937030488] [controller_manager]: Loading controller 'joint_state'
[gzserver-1] [INFO] [1747229499.945253379] [controller_manager]: Loading controller 'mecanum_controller'
[spawner-7] [INFO] [1747229499.954733220] [spawner_joint_state]: Loaded joint_state
[gzserver-1] [INFO] [1747229499.962770650] [controller_manager]: Configuring controller 'joint_state'
[gzserver-1] [INFO] [1747229499.962941166] [joint_state]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-6] [INFO] [1747229499.979108479] [spawner_mecanum_controller]: Loaded mecanum_controller
[gzserver-1] [INFO] [1747229499.981570838] [controller_manager]: Configuring controller 'mecanum_controller'
[gzserver-1] [INFO] [1747229499.985544767] [mecanum_controller]: MecanumDriveController configured successfully
[spawner-7] [INFO] [1747229500.039617591] [spawner_joint_state]: Configured and activated joint_state
[spawner-6] [INFO] [1747229500.078858238] [spawner_mecanum_controller]: Configured and activated mecanum_controller
[INFO] [spawner-7]: process has finished cleanly [pid 10025]
[INFO] [spawner-6]: process has finished cleanly [pid 10023]

```

wahaha-1, 2025-05-16:
I've found the reason: I needed to add a friction direction for each wheel. However, the new problem is that my robot's movement is very unstable; it experiences inexplicable jittering/shaking, whether moving forward or sideways (left/right). Could you provide suitable parameters? Or how should I go about adjusting them?
```
      <geometry>
        <cylinder length="${wheel_link_length}" radius="${wheel_link_radius}"/>
      </geometry>
      <surface>
        <friction>
          <ode>
            <mu>0.05</mu> 
            <mu2>1.2</mu2>
            <fdir1>1 1 0</fdir1>
          </ode>
        </friction>
        <contact>
          <ode>
            <kp>1000000.0</kp>
            <kd>100.0</kd>
            <min_depth>0.001</min_depth>
            <max_vel>0.1</max_vel>
          </ode>
        </contact>
      </surface>
    </collision>
  </link>
```
```
            <joint name="wheel_right_front_joint">
                <command_interface name="velocity">
                    <param name="min">-10</param>
                    <param name="max">10</param>
                </command_interface>
                <command_interface name="effort">
                    <param name="min">-0.1</param>
                    <param name="max">0.1</param>
                </command_interface>
                <state_interface name="position" />
                <state_interface name="velocity" />
                <state_interface name="effort" />
            </joint>
```

christophfroehlich, 2025-05-21:
gazebo classic (and gazebo_ros2_control) reached end-of-life and we can't give support anymore. Please update to gz_ros2_control, having a look at the [released demo](https://control.ros.org/rolling/doc/gz_ros2_control/doc/index.html#mobile-robots) there.

closed 2025-05-21 by christophfroehlich