ros-controls/ros2_controllers #1693: Joint Trajectory Controller is crashing while simulation

vdovetzi, 2025-05-19:
# Description

I am training the RL model of the robotic arm using JTC and gazebo harmonic. I have some hyperparams and one of them is the number of steps in the episode. Basically it is set to 50.

1-2 minutes maximum after the start of the training the warning appear:
```
[gazebo-1] [WARN] [1747657063.981263363] [realtime_tools]: goal_handle attempted invalid transition from state EXECUTING with event CANCELED, at ./src/rcl_action/goal_handle.c:95
```  

It is spamming in the output and then at some random moment this happens:
```
[gazebo-1] Stack trace (most recent call last) in thread 294:
[gazebo-1] #9    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb44857aa33, in __clone
[gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb4484edaa3, in 
[gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fb443779db3, in 
[gazebo-1] #5    Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0x7fb442b26d45, in 
[gazebo-1] #4    Object "/opt/ros/jazzy/lib/libgz_ros2_control-system.so", at 0x7fb419b95428, in gz_ros2_control::GazeboSimROS2ControlPlugin::PostUpdate(gz::sim::v8::UpdateInfo const&, gz::sim::v8::EntityComponentManager const&)
[gazebo-1] #3    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x7fb419aa7be8, in controller_manager::ControllerManager::update(rclcpp::Time const&, rclcpp::Duration const&)
[gazebo-1] #2    Object "/opt/ros/jazzy/lib/libcontroller_interface.so", at 0x7fb41964777a, in controller_interface::ControllerInterfaceBase::trigger_update(rclcpp::Time const&, rclcpp::Duration const&)
[gazebo-1] #1    Object "/opt/ros/jazzy/lib/libjoint_trajectory_controller.so", at 0x7fb40815c4e4, in joint_trajectory_controller::JointTrajectoryController::update(rclcpp::Time const&, rclcpp::Duration const&)
[gazebo-1] #0    Object "/opt/ros/jazzy/lib/libjoint_trajectory_controller.so", at 0x7fb4081624ed, in joint_trajectory_controller::JointTrajectoryController::fill_partial_goal(std::shared_ptr<trajectory_msgs::msg::JointTrajectory_<std::allocator<void> > >) const
[gazebo-1] Segmentation fault (Address not mapped to object [0x1])
[gazebo-1] Segmentation fault (core dumped)
[ERROR] [gazebo-1]: process has died [pid 50, exit code 139, cmd 'ruby $(which gz) sim -r -v 4 --headless-rendering -s /root/sagittarius_ws2/install/sagittarius_gazebo/share/sagittarius_gazebo/worlds/empty.world --force-version 8'].
``` 

Here is my .yaml:

```
controller_manager:
  ros__parameters:
    update_rate: 100 # update_rate specifies how often (in Hz) the controllers should be updated.
 
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    sagittarius_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController
    

    sagittarius_gripper_controller:
      type: joint_trajectory_controller/JointTrajectoryController

sagittarius_arm_controller:
  ros__parameters:
    joints:
      - joint1
      - joint2
      - joint3
      - joint4
      - joint5
      - joint6
    
    # The controller will expect position commands as input for each of these joints.
    command_interfaces:
      - position

    # Tells the controller that it should expect to receive position data as the state
    # feedback from the hardware interface.
    state_interfaces:
      - position

    state_publish_rate: 50.0 # Defaults to 50
    action_monitor_rate: 20.0 # Defaults to 20
    
    # If true, When set to true, the controller will not use any feedback from the system
    # (e.g., joint positions, velocities, efforts) to compute the control commands.
    open_loop_control: false

    # When set to true, it allows the controller to integrate the trajectory goals it receives.
    # This means that if the goal trajectory only specifies positions, the controller will
    # numerically integrate the positions to compute the velocities and accelerations required
    # to follow the trajectory.
    allow_integration_in_goal_trajectories: true

    # Allow non-zero velocity at the end of the trajectory
    allow_nonzero_velocity_at_trajectory_end: true

    allow_partial_joints_goal: true # Defaults to false
    hardware_state_has_offset: true
    deduce_states_from_derivatives: true

    constraints:
      stopped_velocity_tolerance: 0.01 # Defaults to 0.01
      goal_time: 0.0 # Defaults to 0.0 (start immediately)

sagittarius_gripper_controller:
  ros__parameters:
    joints:
      - joint_gripper_left
    
    command_interfaces:
      - position
    
    state_interfaces:
      - position

    state_publish_rate: 50.0 # Defaults to 50
    action_monitor_rate: 20.0 # Defaults to 20
    
    open_loop_control: false

    allow_integration_in_goal_trajectories: true

    allow_nonzero_velocity_at_trajectory_end: true

    allow_partial_joints_goal: true
    hardware_state_has_offset: true
    deduce_states_from_derivatives: true
  
    constraints:
        stopped_velocity_tolerance: 0.01 # Defaults to 0.01
        goal_time: 0.0 # Defaults to 0.0 (start immediately)

```

Here is how I use JTC:

```
def step_arm_request(self, action_values):
        arm_point_msg = JointTrajectoryPoint()
        arm_point_msg.positions = action_values[:DOF]
        arm_point_msg.velocities = [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        arm_point_msg.accelerations = [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        arm_point_msg.time_from_start = GetDurationFromSecs(self.step_time)

        gripper_point_msg = JointTrajectoryPoint()
        gripper_point_msg.positions = action_values[DOF:]
        gripper_point_msg.velocities = [0.0]
        gripper_point_msg.accelerations = [0.0]
        gripper_point_msg.time_from_start = GetDurationFromSecs(self.step_time)

        arm_goal_msg = FollowJointTrajectory.Goal()
        gripper_goal_msg = FollowJointTrajectory.Goal()

        arm_goal_msg.goal_time_tolerance = GetDurationFromSecs(self.step_time)
        gripper_goal_msg.goal_time_tolerance = GetDurationFromSecs(self.step_time)

        arm_goal_msg.trajectory.joint_names = ARM_JOINT_NAMES
        gripper_goal_msg.trajectory.joint_names = GRIPPER_JOINT_NAMES

        arm_goal_msg.trajectory.points = [arm_point_msg]
        gripper_goal_msg.trajectory.points = [gripper_point_msg]

        self.logger.debug("Waiting for action server...")
        self.arm_client.wait_for_server()
        self.gripper_client.wait_for_server()

        self.logger.debug("Sending arm goal-action request...")
        f1 = self.arm_client.send_goal_async(arm_goal_msg)

        self.logger.debug("Sending gripper goal-action request...")
        f2 = self.gripper_client.send_goal_async(gripper_goal_msg)

        rclpy.spin_until_future_complete(self, f1)
        rclpy.spin_until_future_complete(self, f2)
```

# Ideas

I suppose that this issue is about the moment when the first action is interrupted by the next action. I thought this behavior shouldn't be erroneous, but I can be mistaken. Even if so, I think it should not crash like that.

christophfroehlich, 2025-05-19:
This should not happen. So you don't wait for the result of the action, and possibly preempt an active goal, right?
As you are always sending only the next position, why don't you use the simpler trajectory message instead of the action interface?

vdovetzi, 2025-05-20:
Yes, you are right, I will try later today approach with sending command messages via topics. I'll write about results later here, but I still need action interface in `reset` function, because I need to spin until action completes to set the arm in the initial position.

> As you are always sending only the next position, why don't you use the simpler trajectory message instead of the action interface?

vdovetzi, 2025-05-20:
I'm here with results! So, the experiment showed that now it works. 

```
def step_arm_request(self, action_values):
        arm_msg = JointTrajectory(
                joint_names=ARM_JOINT_NAMES,
                points = [JointTrajectoryPoint(
                    positions = action_values[:DOF],
                    time_from_start = GetDurationFromSecs(self.step_time)
                )]
        )

        gripper_msg = JointTrajectory(
                joint_names = GRIPPER_JOINT_NAMES,
                points = [JointTrajectoryPoint(
                    positions = action_values[DOF:],
                    time_from_start = GetDurationFromSecs(self.reset_time)
                )]
        )

        self.arm_pub.publish(arm_msg)
        self.gripper_pub.publish(gripper_msg)
``` 

But is it correct that the JTC's action server cannot withstand the load of sending a bunch of requests with a certain frequency? Do you have such benchmarks or is it not a good idea by design to stress the action server like that?

christophfroehlich, 2025-05-21:
Glad you found a solution. We don't have benchmarks or tests spamming the action interface, it might even be that the action interface of rclcpp is not designed for that. E.g, moveit servo also uses the topic interface of JTC. 

Can we close the issue?

vdovetzi, 2025-05-21:
Yep, the issue can be closed. Thanks!

closed 2025-05-21 by vdovetzi