ros-controls/ros2_controllers #1740: [JSB] Fix reorder warning

christophfroehlich, 2025-06-06:
## Background

```
/workspaces/ros2_rolling_ws/build/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster_parameters.hpp: In constructor ‘joint_state_broadcaster::ParamListener::ParamListener(const std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>&, rclcpp::Logger, const std::string&)’:
/workspaces/ros2_rolling_ws/build/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster_parameters.hpp:316:22: warning: ‘joint_state_broadcaster::ParamListener::logger_’ will be initialized after [-Wreorder]
  316 |       rclcpp::Logger logger_;
      |                      ^~~~~~~
/workspaces/ros2_rolling_ws/build/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster_parameters.hpp:309:19: warning:   ‘std::string joint_state_broadcaster::ParamListener::prefix_’ [-Wreorder]
  309 |       std::string prefix_;
      |                   ^~~~~~~
/workspaces/ros2_rolling_ws/build/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster_parameters.hpp:103:5: warning:   when initialized here [-Wreorder]
  103 |     ParamListener(const std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>& parameters_interface,
      |     ^~~~~~~~~~~~~
```

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

dimitridecious, 2025-06-09:
Hey, I'm working on fixing this. I can see the problem is in the **auto-generated** parameter header file, which comes from the generate_parameter_library template.
Quick question - where should this actually be fixed? Should I be submitting a PR to the PickNikRobotics/generate_parameter_library repo to fix their Jinja template, or is there a local workaround you had in mind for the ros2_controllers repository? Since this is marked as a "good first issue" I want to make sure I'm taking the right approach before diving into modifying system templates.
The fix itself is straightforward (changing constructor assignments to member initialization list), just want to confirm the right place to implement it.
Thanks!

christophfroehlich, 2025-06-09:
I haven't thought about it but just copied the warning into a good-first issue ;) If you have found the library to be the culprit, please open a PR there!

christophfroehlich, 2025-07-26:
Already fixed, thanks!

closed 2025-07-26 by christophfroehlich