ros-controls/ros2_controllers #1757: Proposal: Modular Reference Handling in ROS 2 Controllers

my-rice, 2025-06-13:
Hi, we are writing this issue as part of the [Automatic Control Group from the University of Salerno](https://github.com/unisa-acg). We have previously contributed to the ROS ecosystem, for example, through pull requests such as [here](https://github.com/ros-controls/kinematics_interface/pull/92) and [here](https://github.com/ros-controls/kinematics_interface/pull/93), as well as by deploying ROS 2 packages to the community like [here](https://index.ros.org/p/inverse_dynamics_solver/).

---
We are opening this issue to propose a new design approach for the control framework, specifically aimed at improving how controllers handle reference inputs.

**Is your feature request related to a problem? Please describe.**
After reviewing the current implementations, we observed that a significant portion of controller code is dedicated to managing reference acquisition, typically via subscribers or action servers. For example, [here](https://github.com/ros-controls/ros2_controllers/tree/master/admittance_controller) and [here](https://github.com/ros-controls/ros2_controllers/tree/master/joint_trajectory_controller).

**Describe the solution you'd like**
To improve modularity and separation of concerns, we propose a cleaner structure that decouples reference acquisition from control logic. The idea is to divide responsibilities into two distinct modules:

1. **Reference Generator** – This component is practically a controller, but its purpose is to receive and validate reference trajectories or setpoints from the ROS 2 network. It abstracts the communication interface and provides a clean output to downstream controllers, using [controller chaining](https://control.ros.org/rolling/doc/ros2_control/controller_manager/doc/controller_chaining.html).

2. **Controller** – This component is exclusively responsible for computing the control law. It receives input from the Reference Generator and remains agnostic to how references are produced or delivered.

This modular structure promotes **better organization and a clearer separation of responsibilities**. Moreover, Reference Generators **increase code reuse** across multiple controllers. The diagram below illustrates the proposed architecture:
[image]
Importantly, **this proposal does not require any modifications to existing controllers**. The framework is fully backward-compatible, allowing existing controllers to continue functioning without changes. Developers creating new controllers may adopt this structure if it aligns with their needs.

**Describe alternatives you've considered**
The most straightforward alternative is to maintain the current design, leaving users responsible for implementing their own reference-handling logic, even when this functionality could be modularized and reused across controllers.

**Additional context**
We have already internally implemented a prototype of this modular framework and developed demo controllers to demonstrate its benefits. We are happy to make the repository public on request.

We are sharing this issue now to **gather feedbacks** and assess whether this direction aligns with the needs and preferences of the community. We are **happy to discuss, refine, or adjust the approach** based on your input.

eferre89, 2025-07-02:
Dear @destogl,

thank you for your feedback! Indeed, what you are proposing sounds like a third way to us, that is embedding the communication interfaces (action server/subscriber on specific messages) within external libraries, if I caught your point. But I'd like to go to the very origin of our proposal: I guess there is some need, in ros-controls, to deal with
* task-space control (not just performing IK, but I mean defining a task-error-based controller, like task-space inverse dynamics control, for instance).
* interaction control (compliance control, impedance control, admittance control, direct force control, hybrid force/position control; the latter -might- require the definition of a reference wrench)

Clearly, if a planning system was able to generate plans for such controllers, it would likely use different messages, that would not comply with the `JointTrajectory` we are mostly using in these days in ros-controls. This would call for specialized controllers (the reference generators @my-rice was proposing) or specialized libraries to configure the JTC with (perhaps, `JointTrajectoryController` shall become `TrajectoryController` in this case, to accommodate the use cases above).

On the other hand, what is nice about JTC (that I really consider a masterpiece controller) is that it already implements both the PointReference and TrajectoryReference interfaces @my-rice was pointing out in his design through the usage of the same message, i.e. the `JointTrajectory`, and, at the same time, can be chained with arbitrary -joint space- controllers. For instance, we have been validating JTC with both ros-control's `PidController` and a `GravityCompensationPDController` we developed. Both configurations worked nicely.

Therefore, to sum up, in the next future, I would
* keep the JTC as it is, with its ability to chain other controllers and manage both single-point and trajectory references
* place other "controllers" next to it to manage different messages and enable the use cases I mentioned before (these would be the reference generators @my-rice was proposing)

More on the long run, I'd try to
* assess whether controller-level configurability (through either plugin or toolbox library), i.e. same "ROS-communication interface controller" with different configuration to cope with different messages, would be preferable over controller-manager-level configurability, i.e. one controller per message type. I guess this really gets down to the point of whether different messages shall undergo the same logic within the controller. I have ideas about this, but not a definitive answer and deeper analysis would be needed.

For the time being, we have code implementing what @my-rice proposed:
* A joint space reference generator able to accept "extended" joint space references (see https://github.com/ros-controls/control_msgs/pull/221)
* A task space reference generator able to accept task space messages (either trajectories or single points)
* The common logic is included in a superclass the two controllers inherit from, while message-specific processing and input/output mapping (through semantic components) is managed in the derived classes

my-rice, 2025-07-11:
In light of the discussion so far, we’d like to clarify the idea and how it can be unified with the JTC. The idea is about a common base class, say `ReferenceGeneratorBase`, which extends `ControllerInterface` and is designed to group all the common logic, for example, the logic to handle a generic trajectory. 

This base class is meant to be extended by specialized reference generators that focus on different types of reference inputs. For example, `JointWrenchReferenceGenerator` for controllers that work with desired “joint position reference + a task space wrench” trajectories, or a `TaskSpaceReferenceGenerator` for task space messages. Following the same reasoning, the `JTC` would conceptually remain as-is but can be viewed as one of several implementations of the base interface that handles joint trajectory points.

Each of these specialized generators would override the base functionality and adapt it to the message type it handles, maintaining the ROS 2 subscription or action interface inside the generator itself, rather than repeating this logic across controllers. A sketch of this idea is shown below:

[image]

To further explain the applicability of this architecture, here’s a set of meaningful use cases we’ve already explored. Each reference generator can be chained to a wide variety of downstream controllers, depending on the application. In the following image there are some examples:

[image]

For instance, the `JointWrenchReferenceGenerator` can feed an admittance controller. The `TaskSpaceReferenceGenerator` can be used with several kinds of controllers, from simple pose tracking controllers (e.g. Cartesian pose controller) to more complex force-based controllers such as admittance, direct force, impedance and stiffness controllers. Finally, the `JTC` could feed controllers that rely on the information contained in the Joint Trajectory point such as a PID or a gravity compensation controller.

---

Ideally, in the future, the trajectory management logic that is now present in the `JTC` (including interpolation) could be moved to the `ReferenceGeneratorBase` class. However, this requires a careful separate study to understand whether its functionality is independent of the message type. Nevertheless, for the time being, `JTC` can be one of the options to handle specific `JointTrajectory`messages, and the introduction of reference generators (base + derived classes) won't break the current implementation of the `JTC`.

my-rice, 2025-10-07:
Hi everyone,
 
As a follow-up to our previous discussion, I’d like to share [a few demos](https://github.com/unisa-acg/reference-generators-demo) that test both the joint space and task space reference generators in different configurations, using the UR10 robot in simulation. These demos show many chaining combinations between reference generators and downstream controllers.
 
At the moment, the core logic for trajectory handling and validation is reused through a reference generator base class, while the handling logic related to specific messages remains encapsulated in the derived reference generators.
 
We believe these examples could be very useful in designing a solution that could be integrated into the `ros2_control` framework.